roleoroleo/yi-hack-Allwinner #1: Need help - cannot see putty output after soldering

bytesex, 2020-04-11:
Hello there)
I have the Yi 1080p Home camera with a firmware of 8.2.0.0A_201912121707

I did the first step of camera rooting
I also managed to solder wires to Rx Tx and GND pads. What port settings shall I use for serial connect?
Currently I use the speed 9600 and see nothing in putty, having the SD card with 1step' files on it
If the following does matter - I use the DSD TECH USB to TTL Serial Converter
https://www.amazon.com/gp/product/B072K3Z3TL

aholza, 2020-04-12:
Hi,

did you switch rx and tx? Also did you set bitrate to 115200?

bytesex, 2020-04-12:
Yes, the issue is gone, thx - another TTL adapter worked.
Now I get the "syntax error" output from camera:

0.617863] sunxi-mmc sdc1: smc 1 p1 err, cmd 1, RTO !!
starting pid 471, tty '': '/etc/init.d/rcS boot'
[RCS]: /etc/init.d/S01udev
Populating /dev using udev: [    0.925813] udevd[476]: could not create /run/udev: No such file or directory
done
[RCS]: /etc/init.d/S02app
[RCS]: /etc/init.d/S20urandom
--------------------------home app init.sh--------------------------
read-only file system detected...done
starting pid 487, tty '/dev/console': '-/bin/sh'
vm.dirty_background_ratio = 2

BusyBox v1.27.2 () built-in shell (ash)

vm.dirty_expire_centisecs = 500
vm.dirty_ratio = 2
vm.dirty_writeback_centisecs = 100
root@xiaoyi:/# fs.mqueue.msg_max = 256
checkdisk.c(get_tfcard_size_v2-105)[03:04:58.783]:TF card total size : 3781 MB
checkdisk.c(get_fs_type-262)[03:04:58.785]:FS TYPE ERROR!
checkdisk.c(main-400)[03:04:58.785]:FS format error.
rmmod: can't unload module 'exfat': No such file or directory
checkdisk, main() ret_status(10002)
checkdisk, main() ret_status(10002),g_sysconf_ptr->tf_status.stat(10002)
umount: can't unmount /tmp/sd: Invalid argument
---update file(home_y20gam) Not exist---
--------------------------insmod sound--------------------------
[    3.302508] --->sunxi_daudio_dev_probe<--- strat.
[    3.308236] --->sunxi_daudio_dev_probe<--- end.
--------------------------insmod sensor--------------------------
[    3.937033] sunxi-mmc sdc1: smc 1 p1 err, cmd 8, RTO !!
############## Starting Hack ##############
############## Starting Hack ##############
############## Starting Hack ##############
############## Starting Hack ##############
: not foundtory/config.sh: line 5:
: not foundtory/config.sh: line 6:
### Hacking telnet
**/tmp/sd/Factory/config.sh: line 42: syntax error: unexpected end of file (expecting "then")**
>>>>>>>>>>>>>#########is_sd_exit chg to: 10002<<<<<<<<<<<<<<<<<<<Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory
Failed to connect to non-global ctrl_ifname: wlan0  error: No such file or directory

roleoroleo, 2020-04-12:
I tested it with my cam.
Check if it's a linux file (correct end of line).

bytesex, 2020-04-12:
UNIX LF-style ending did the trick, thx

closed 2020-04-12 by bytesex

aholza, 2020-04-12:
Maybe adding a gitattribute file to control line endings for sh scripts.
`*.sh text eol=lf`
if I am not mistaken.