roleoroleo/yi-hack-Allwinner #2: How to compile firmware

aholza, 2020-04-12:
While trying to compile things I did notice that a special SDK is needed to compile thing. compile.sh references /opt/yi/toolchain-sunxi-musl/toolchain, but I cannot really find where I could get this toolchain. Can someone give a hint where to get this?

roleoroleo, 2020-04-12:
Dowload this:
https://github.com/lindenis-org/lindenis-v536-prebuilt/tree/master/gcc/linux-x86/arm/toolchain-sunxi-musl

roleoroleo, 2020-04-18:
I created a wiki page:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Cross-toolchain

gauthier12, 2020-05-12:
I would be willing to have a look but is there a documentation about how to compile correctly and install a modified firmware ? Is-it just the compile.sh script the pack_fw.sh then install it as a release ? Is there a risk to brick the cam if there is a bug in the modified firmware ?

roleoroleo, 2020-05-12:
If you study a few hours the source code you can understand the structure of this project.
For example.
1. To add a program to the fw, simply add a new folder in /src with script to init, compile and install (you can copy it from another program).
2.  To add a new page to the web page add the html, the js and the sh (cgi-bin server side).
3. To add command to busybox, simply edit the .config file.
4. etc...

If you have know-how about cross-compilint, it's very simple.
Then, compile.sh and pack_fw.all.sh

Yes, there is a risk to brick the cam if you overwrite the original files.
In theory, you don't need to overwrite the yi files if the cam is already hacked.
There is a boot-loop risk if something causes a kernel panic (for example) but you can recover this situation reflashing another fw.
Finally you have the serial cable option if the cam is really bricked (I have never needed to use this option).

closed 2020-07-14 by roleoroleo