roleoroleo/yi-hack-Allwinner #3: MQTT Motion Not Connecting to MQ Broker

crdnl, 2020-04-18:
Yi-Hack-Allwinner Version: 0.1.2

## Description:
Using the web interface to configure the MQTT then rebooting the camera does not show that the camera is connecting to the MQ Broker.

In this case, I'm using Mosquitto MQ running on a Raspberry Pi under Docker. Connecting to port 1883.

Checking the Mosquitto logs, I don't see a connection attempt to the broker.

roleoroleo, 2020-04-18:
Log in with ssh and run the program manually:
```
root@yi-hack:~# killall mqttv4
root@yi-hack:~# LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/yi-hack/lib
root@yi-2614:~# /home/yi-hack/bin/mqttv4
```

You should see a log like this:
```
Starting mqttv4 v0.0.3
Trying to connect... Client yi-cam-a sending CONNECT
Client yi-cam-a received CONNACK (0)
. connected!
CALLBACK MOTION STOP
Client yi-cam-a sending PUBLISH (d0, q1, r0, m1, 'home-assistant/cameretta/motion_detection', ... (3 bytes))
Client yi-cam-a received PUBACK (Mid: 1)
CALLBACK MOTION START
Client yi-cam-a sending PUBLISH (d0, q1, r0, m2, 'home-assistant/cameretta/motion_detection', ... (2 bytes))
Client yi-cam-a received PUBACK (Mid: 2)
```

Post your log.

roleoroleo, 2020-04-18:
Check if the client id is correct.
If you have more than 1 cam you must use different names otherwise the server will drop the connections.

crdnl, 2020-04-18:
```

root@xiaoyi:~# killall mqttv4
killall: mqttv4: no process killed
root@xiaoyi:~# LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/yi-hack/lib
root@xiaoyi:~# /home/yi-hack/bin/mqttv4
Starting mqttv4 v0.0.3
Can't open file "/home/yi-hack-v4/etc/mqttv4.conf": No such file or directory
Cannot open config file. Skipping.
Trying to connect... Unable to connect (Connection refused).
Unable to connect (Connection refused).
Unable to connect (Connection refused).
Unable to connect (Connection refused).
Unable to connect (Connection refused).
Unable to connect (Connection refused).
```

As discussed in Discord, it looks like a missing space in the setup script. There's no symlink between `yi-hack-v4` and `yi-hack`.

roleoroleo, 2020-04-18:
Fixed here:
https://github.com/roleoroleo/yi-hack-Allwinner/commit/0f22892df1b9ad2e0ed80f1115df70d148b68b79

crdnl, 2020-04-27:
Issue fixed in 0f22892, closing issue.

closed 2020-04-27 by crdnl