roleoroleo/yi-hack-Allwinner #5: Snapshot vs out of memory

roleoroleo, 2020-04-18:
This cam works in a very critical memory situation.
```
root@xiaoyi:~# /home/yi-hack/usr/bin/free
              total        used        free      shared  buff/cache   available
Mem:          61164       45520        3428        1908       12216        7444
Swap:             0           0           0
```
During normal operations the available memory is more or less 3 MB and the main process (rmm) occupies more than 60% of it.
Unfortunately, it is frequent to have out of memory problems so it is very difficult to develop new features.
For example, a malloc of a yuv image (3110400 bytes) is enough to have problems.
If you take a snapshot sequence the cam will restart.

crdnl, 2020-04-18:
Can we cover what `rmm` is required for? Is it essentially everything?

roleoroleo, 2020-04-18:
If we want to remove rmm, we have to rewrite video acquisition and encoding.
We would necessarily need an sdk.

gauthier12, 2020-04-19:
Sorry if it is a stupid question but isn'it possible to activate swap ? It maybe very slow may at least it won't be out of memroy

roleoroleo, 2020-04-19:
Yes, it is possible to activate the swap (even if swapping to SD is not a smart idea) but this means that in order for the hack to work it is necessary to have an sd.
I do not really like, but...

crdnl, 2020-04-19:
I'll take some time to did through rmm and maybe we can just disable portions of it.

roleoroleo, 2020-04-29:
I tested the system by disabling the cloud (and therefore several programs that don't start) and it seems to work much better.
Taking a sequence of snapshot does not restart.

Xandrios, 2020-05-12:
With the cloud disabled (as well as all other services except HTTP and SSH) it does restart for me. The first snapshot succeeds, the second causes a restart. 

What I do wonder though, is why a subsequent run of the snapshot causes such issue. When the first run has completed all memory used for that request should have been free'ed right? Meaning that all memory used should have been returned and available for a second snapshot run. If a subsequent run causes an out-of-memory, does that not mean that we have a memory leak in that service somehow?

In any case I'd very much like to be able to use SWAP when an SD-card is available. Maybe we can make it conditional? If SD is available enable SWAP, if no card, leave it disabled. That way we can at least prevent crashes while executing short-lived memory-intensive tasks.

Personally I'd always be using an SD card anyway, and since these things cost next to nothing there should not really be a reason to not have one right? An 8 or 16GB card should be only a few bucks.

gauthier12, 2020-05-12:
Furthermore, an SD card is need to flash the firmware.

roleoroleo, 2020-05-12:
Ok, added.

Xandrios, 2020-05-12:
Much appreciated. I've ran a quick test and it seems to work:

1. Prepared a 1GB swap file on another *nix box (link [here](http://hetz1.xandrios.net/swapfile.gz), just unzip for use)
2. Issue command to activate swap: swapon /tmp/sd/swapfile

We don't have many tools to check swap usage, but this seems to work:

`root@Testcam1:~# cat /proc/swaps
Filename                                Type            Size    Used    Priority
/tmp/sd/swapfile                        file            1048572 4888    -1`

Now we only have to make that swapon command be executed at startup IF there is a swapfile available on the SD card. What's the 'best practice' to execute scripts at startup?

roleoroleo, 2020-05-12:
Already done.
https://github.com/roleoroleo/yi-hack-Allwinner/commit/7b000f84034adeb62e2d042c4b795c945be256e6

Xandrios, 2020-05-12:
Wow, you are very fast! Awesome.

I didn't have the mkswap available hence I created the swapfile on another platform. But CONFIG_MKSWAP will take care of that likely. Thanks!

roleoroleo, 2020-05-12:
Yes, I released 0.1.6 with this feature.

Xandrios, 2020-05-12:
Just upgraded and tested: Works like a charm! Much appriciated.

closed 2020-07-14 by roleoroleo