roleoroleo/yi-hack-Allwinner #7: MQTT "Cannot open directory"

TimDowker, 2020-04-23:
Trying to work out the mqtt-home assistant stuff.  I have other MQTT devices around my house all working as expected.

After activating MQTT on the camera I wasn't receiving the motion_start or motion_stop signals.

Found this

[https://github.com/roleoroleo/yi-hack-Allwinner/issues/3](https://github.com/roleoroleo/yi-hack-Allwinner/issues/3)

and did the suggested

root@yi-hack:~# killall mqttv4
root@yi-hack:~# LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/yi-hack/lib
root@yi-hack:~# /home/yi-hack/bin/mqttv4

Now I get this on screen

CALLBACK MOTION START
Client yi-cam sending PUBLISH (d0, q1, r1, m16, 'yicam/motion', ... (12 bytes))
Client yi-cam received PUBACK (Mid: 15)
Client yi-cam received PUBACK (Mid: 16)
SENDING FILES LIST
Cannot open directory: 2020Y04M23D12H
Client yi-cam sending PUBLISH (d0, q1, r1, m17, 'yicam/files', ... (85 bytes))
Client yi-cam received PUBACK (Mid: 17)
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r1, m18, 'yicam/motion', ... (11 bytes))
CALLBACK MOTION START
Client yi-cam sending PUBLISH (d0, q1, r1, m19, 'yicam/motion', ... (12 bytes))
Client yi-cam received PUBACK (Mid: 18)
Client yi-cam received PUBACK (Mid: 19)
SENDING FILES LIST
Cannot open directory: 2020Y04M23D12H

Wondering if the "Cannot open directory" is what's causing the problems.

Thank you for any help you can provide.

TimDowker, 2020-04-23:
I have both FTP services disabled as well to save on resources.  Do they need to be enabled to have MQTT function properly?

TimDowker, 2020-04-23:
For reference - I used the bottom instructions of this page to setup the binary sensor in Home Assistant

[https://github.com/TheCrypt0/yi-hack-v4/wiki/Home-Assistant-RTSP-configuration](https://github.com/TheCrypt0/yi-hack-v4/wiki/Home-Assistant-RTSP-configuration)

roleoroleo, 2020-04-23:
Check your server config.
The log seems to be correct:
```
Client yi-cam sending PUBLISH (d0, q1, r1, m18, 'yicam/motion', ... (11 bytes))
```