roleoroleo/yi-hack-Allwinner #8: Cannot set timezone - Read-only file system

TimDowker, 2020-04-23:
Trying to set the appropriate timezone and receive this error after trying to run the command 

`echo TimeZone > /etc/TZ`

-ash: can't create /etc/TZ: Read-only file system

Found at 
[https://github.com/TheCrypt0/yi-hack-v4/wiki/Setting-the-Timezone](https://github.com/TheCrypt0/yi-hack-v4/wiki/Setting-the-Timezone)

roleoroleo, 2020-04-23:
The rootfs is read only.
It's different from the previous cams.
To solve this issue we should rewrite the rootfs increasing the risk of brick.

gmarfjan, 2020-04-23:
I've done an ugly fix in /home/yi-hack/script/system.sh by adding `export TZ="GMT-2"` before starting all services (GMT-2 is my timezone). So all process start with this environment variable. Now watermark on snapshot displays the right date. When I'm connected by ssh, I have a wrong date because TZ is not set (should be set in /etc/profile but the fs is read-only). When I set it manually, it looks good.
We could add TZ parameter in /home/yi-hack/etc/system.conf and add `export TZ=$(get_config TZ)` in  /home/yi-hack/script/system.sh . Web UI should be modified by adding a droplist of all timezones.
Do you think that's a good way ?

roleoroleo, 2020-04-24:
In this maner, only "hacked" services start with new TZ.

Yi services start without it.

If it is enough it could be a good solution

gmarfjan, 2020-04-24:
That's not the right way. For completely resolve this issue, I agree with you, we have to fix the TZ on OS level. It's just a temporary fix.

TimDowker, 2020-04-26:
Thank you for adding the setting in the config.  However it doesn't seem to be retaining the setting.

roleoroleo, 2020-04-26:
I retested it now and it seems to work.

TimDowker, 2020-04-26:
What timezone are you using?  I'm trying America/Toronto which is 

"EST5EDT,M3.2.0,M11.1.0"

And it doesn't work.  Also tried just "EST5EDT" and same thing.

roleoroleo, 2020-04-26:
CET-1CEST,M3.5.0,M10.5.0/3

TimDowker, 2020-04-26:
Still no go.  What can I do to help diagnose this?

roleoroleo, 2020-04-26:
Check if the file /home/yi-hack/etc/system.conf contains the item "TIMEZONE".

TimDowker, 2020-04-26:
It does.

`root@yi-b73d:/home/yi-hack/etc# cat system.conf 
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=yes
RTSP=yes
RTSP_STREAM=both
ONVIF=no
ONVIF_PROFILE=both
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
RTSP_PORT=554
RTSP1_PORT=8554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
FREE_SPACE=10
ONVIF_WM_SNAPSHOT=no
TIMEZONE=
root@yi-b73d:/home/yi-hack/etc#`

roleoroleo, 2020-04-26:
Try to add a new line after TIMEZONE=

TimDowker, 2020-04-26:
Ok, so adding the line didn't work but I was able to manually add it to system.conf.  It still doesn't show anything in the "Timezone" text box on the Configurations page however the correct time is shown on the main information page.

TimDowker, 2020-04-26:
`date` doesn't return the correct date though.

roleoroleo, 2020-04-26:
I can't understand...

roleoroleo, 2020-04-26:
I restored 0.1.3 and upgraded again to 0.1.4.
I have no problems.

roleoroleo, 2020-04-29:
Has anyone tested this functionality with version 0.1.4?

closed 2020-04-29 by TimDowker

emeins, 2020-04-30:
I tested with version 1.4.0. I added the TZ string (CET-1CEST,M3.5.0,M10.5.0/3) in the general settings of the web interface and it works well. I can see the TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3 in the /home/yi-hack/etc/system.conf without adding additional lines manually. I was a bit confused when I checked with #date command in ssh shell - still GMT time. But the snapshot and motion events present the right timezone.

roleoroleo, 2020-04-30:
In shell you can't have the variable because there isn't profile file.