roleoroleo/yi-hack-Allwinner #9: Can't flash camera

MrPatate87, 2020-04-23:
I try to flash BFUS Yi Camera under 8.2.0.0A firmware with Allwinner hack, put file on the root of my sd card (try 3 micro sd 8G x 2 and 2G) but every time th camera boot directly and don't flash new fimware.
I've already try MStar verqion of the hack but I ind in the GIT issue this firmware was not supported.
I try many things to flash this camera but can't.

scruff-01, 2020-04-23:
Everything in the tgz file goes on the sd card. You said file but there is more than one

MrPatate87, 2020-04-24:
my fault. I put home_y20gam.stage file and the 2 folder in the micro sd (formated in Fat32 cause if i format the card in the camera I can't read it on my pc), but when I boot the camera with the miscro sd he camera simply boot and nothing more happen.

scruff-01, 2020-04-24:
Strange. It can take a bit for it to update. When you remove the sd card after trying to update is there an extra folder on the sd card called record? Just wondering if it has problem reading or writing to the card

MrPatate87, 2020-04-25:
Yes i can see 2 folders created by thé camera, record and log, try another time this morning, access to thé micro sd card with my android Phone, don't format it (already done by the camera a couple of time before), i copy past the yi hack allwinner files on it (don't remove record folder and log).
Reboot the camera with this sdcard and it's done, work fine. It take a couple of minute to access to the config but it's OK.
Thanks for your help and many Thanks for this great hack !

closed 2020-04-25 by roleoroleo