roleoroleo/yi-hack-Allwinner #10: Home Assistant Integration - (Rewriting root fsto make it read/write)

justinhelmer, 2020-04-26:
Hello @roleoroleo,

First I want to say a huge thanks for putting this together and maintaining it. I was able to get 4x BFUS* cameras flashed with the custom firmware, and running RTSP/ONVIF with very little effort thanks to all the hard work. Everything works smoothly (other than the known [stream artifact](https://github.com/roleoroleo/yi-hack-Allwinner/issues/4) issue).

I am using the wildly popular [Home Assistant](https://www.home-assistant.io/) platorm, and was able to get the snapshots & streams working through the platform using the [ONVIF](https://www.home-assistant.io/integrations/onvif/) component. Awesome.

---

**The Goal**

Although ONVIF is useful, direct integration with the Yi Camera platform is much preferred, for several reasons.

One main reason is so the binary sensors are registered, allowing use of the cameras as occupancy/motion sensors. This enables running automated routines based on these triggers (i.e. in-home security system triggers, turning lights on automatically, etc.)

Additionally, using the [Yi Camera component](https://www.home-assistant.io/integrations/yi/) would also theoretically allow access the direct camera stream, avoiding the need to use RTSP/ONVIF. This would also circumvent the current issues with [distortion](https://github.com/roleoroleo/yi-hack-Allwinner/issues/4).

**The Problem**

The [Home Assistant Integration](https://www.home-assistant.io/integrations/yi/) requires the FTP server to be enabled, with a password set. Without a password set, it will not work. However, running the `passwd` command ultimately results in the following failure:

```
passwd: /etc/passwd: Read-only file system
passwd: can't update password file /etc/passwd
```

As you referenced in [another issue](https://github.com/roleoroleo/yi-hack-Allwinner/issues/8), the rootfs is readonly for the new cameras. However, this essentially means that it is not possible to integrate these newer cameras with Home Assistant using the [Yi Camera component](https://www.home-assistant.io/integrations/yi/).

Is there any chance that either yourself or the community are willing to pursue this further? Home Assistant integration would be incredibly useful.

Thanks for your time.

roleoroleo, 2020-04-26:
I didn't try ftp component in hass but if you use a random password with root user you should log in.
I tested it with filezilla and with a command line client.

Otherwise we could modify the component to use the getlastrecordedvideo.sh API made by @agsola.

justinhelmer, 2020-04-26:
Good thought; I didn't realize FTP behaved this way. Although less-than-secure, I tried this in the interest of debugging. I also noticed the Home Assistant component config for [password](https://www.home-assistant.io/integrations/yi/#password) says

>"Can be any string as the current firmware doesn’t allow setting FTP passwords."

(contradictory to an earlier statement made in the same docs).

I was also able to connect via an FTP client like yourself (as a control test), but it appears the Yi camera component in Hass is not happy.

I SSH'd in to the camera to try to discover the record path, which appears to be different than the HA component's default. Using this config:

```
camera:
  - platform: yi
    name: Yi Cam Test
    host: 192.168.1.125
    password: randomdoesntmatter
    path: /tmp/sd/record
```

Home Assistant makes the following request:

```
http://192.168.1.3:8123/api/camera_proxy/camera.yi_cam_test?authSig=<auth>
```

...and gets back:

```
500 Internal Server Error
Server got itself in trouble
```

This is what show up in the HA logs:

```
2020-04-25 23:21:19 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_protocol.py", line 418, in start
    resp = await task
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_app.py", line 458, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_middlewares.py", line 119, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/real_ip.py", line 39, in real_ip_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 72, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 127, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 123, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/camera/__init__.py", line 492, in get
    return await self.handle(request, camera)
  File "/usr/src/homeassistant/homeassistant/components/camera/__init__.py", line 509, in handle
    image = await camera.async_camera_image()
  File "/usr/src/homeassistant/homeassistant/components/yi/camera.py", line 129, in async_camera_image
    url = await self._get_latest_video_url()
  File "/usr/src/homeassistant/homeassistant/components/yi/camera.py", line 101, in _get_latest_video_url
    latest_dir = dirs[-1]
IndexError: list index out of range
```

I tried with both standard and legacy FTP. Same error in both scenarios.

I know this is not a "Home Assistant" repo, so I'm not sure if you have any suggestions from here. Unfortunately I don't have the knowledge needed to contribute to either HA or the yi-hack* projects.

Even if all of the work were done to enable a R/W rootfs, its possible the component would still have issues since I don't have a clean blue/green read because I'm trying to get the component set up for the first time and I don't have any older Yi cams to test configuration with to create a control group.

Any idea how to move forward from here?

roleoroleo, 2020-04-26:
I think the hass component is working with the yi cam models.
I don't know if the directories/files structure is the same as the new models.
I will investigate.

justinhelmer, 2020-04-27:
Edited title: Removed statement about assumption related to solution being readonly rootfs

roleoroleo, 2020-04-27:
I didn't try the hass component but I tried get_latest_video_url function:
https://github.com/home-assistant/core/blob/dev/homeassistant/components/yi/camera.py
and it works.

The error "latest_dir = dirs[-1]" probably occurs when there are no video folders in the /tmp/sd/record folder.
Could you check it?
And try this:
1 - remove the content of the folder /tmp/sd/record
2 - wait for a motion event
3 - check if the folder contains a new directory
4 - check if the directory contains a new mp4 file
5 - run the script again and check the log

justinhelmer, 2020-04-27:
Making progress I think (new error):

```
2020-04-27 15:38:02 ERROR (MainThread) [aiohttp.server] Error handling request
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/yi/camera.py", line 90, in _get_latest_video_url
    await ftp.connect(self.host)
  File "/usr/local/lib/python3.7/site-packages/aioftp/client.py", line 519, in connect
    await super().connect(host, port)
  File "/usr/local/lib/python3.7/site-packages/aioftp/client.py", line 137, in connect
    self.ssl,
  File "/usr/local/lib/python3.7/site-packages/aioftp/client.py", line 39, in open_connection
    host, port, ssl=ssl)
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 962, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 949, in create_connection
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.7/asyncio/selector_events.py", line 473, in sock_connect
    return await fut
  File "/usr/local/lib/python3.7/asyncio/selector_events.py", line 503, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('192.168.1.125', 21)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_protocol.py", line 418, in start
    resp = await task
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_app.py", line 458, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/web_middlewares.py", line 119, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/real_ip.py", line 39, in real_ip_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 72, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 127, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 123, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/camera/__init__.py", line 492, in get
    return await self.handle(request, camera)
  File "/usr/src/homeassistant/homeassistant/components/camera/__init__.py", line 509, in handle
    image = await camera.async_camera_image()
  File "/usr/src/homeassistant/homeassistant/components/yi/camera.py", line 129, in async_camera_image
    url = await self._get_latest_video_url()
  File "/usr/src/homeassistant/homeassistant/components/yi/camera.py", line 93, in _get_latest_video_url
    raise PlatformNotReady(err)
homeassistant.exceptions.PlatformNotReady: [Errno 111] Connect call failed ('192.168.1.125', 21)
```

Also, it is worth noting the request is for the _thumbnail_, not the video feed itself.

To be explicit, the thumbnail request:

```
http://192.168.1.3:8123/api/camera_proxy/camera.yi_cam_test?authSig=<authSig>
```

....and its response:

```
500 Internal Server Error
Server got itself in trouble
```
> HTTP 500

(Same as original description)

It is the **thumbnail** request that generates the error log above (and in my original post).

When I click on the component to view the feed, this is the request:

```
http://192.168.1.3:8123/api/camera_proxy_stream/camera.yi_cam_test?token=<token>
```

It responds immediately with a `200` and `multipart/x-mixed-replace;boundary=ffmpeg`, but the network request fails in the browser: `(failed)net::ERR_INCOMPLETE_CHUNKED_ENCODING`.

I would assume the error log with the _thumbnail_ request points to the same issue that will solve both cases.

justinhelmer, 2020-04-28:
Update:

I had forgotten to re-enable the legacy FTP server as part of this follow-up test. After doing so I am 
presented with the same error as in the original description :-( 

```
// ... (rest omitted for brevity)

File "/usr/src/homeassistant/homeassistant/components/yi/camera.py", line 101, in _get_latest_video_url
    latest_dir = dirs[-1]
IndexError: list index out of range
```

Seems like your suggestion unfortunately did not work. Other ideas?

Perhaps @frenck or @balloob may have some suggestions - they both seem to have been major contributors to the Yi Component

frenck, 2020-04-28:
You have looked wrongly. Thanks for the random mention.

justinhelmer, 2020-04-28:
@frenck - apologies; was basing it off the [commit history](https://github.com/home-assistant/core/commits/dev/homeassistant/components/yi) for the component, without digging into the contents of the commits themselves. Carry on

roleoroleo, 2020-04-28:
I studied the yi component and I had several doubts.
The code retrieves the last video recorded via ftp, this means:
1. You have to configure the cam to always save videos (no motion detection).
2. When you get a stream or an image you are about 2 minutes late.
3. The videos are only 1 minute long.

I think it is not convenient to use this component compared to onvif/rtsp.
Regarding motion, you can use the mqtt feature to get motion detection events.

justinhelmer, 2020-05-01:
Makes sense to me.

closed 2020-05-01 by justinhelmer

Pe-MaKer, 2020-07-31:
> I think it is not convenient to use this component compared to onvif/rtsp.
> Regarding motion, you can use the mqtt feature to get motion detection events.

Thanks for this firmware hack @roleoroleo .
I have integrated the Yi Home into Home Assistant (0.113.2) via ONVIF and MQTT as recommended.
Unfortunately only the one camera stream is recognized by ONVIF, but no sensors, switches or snapshots. The camera is found via ONVIF Discovery, but not all attributes may be negotiated. And also via MQTT I can only detect movements and Baby Cry.

I am very spoiled by "Dafang Hacks" [https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/integration/homeassistant/homeassistant.md]
where almost all functions can be accessed via MQTT.

I can live with all this and I am happy to be free from the Yi-Cloud.
But what for me is a real limitation in usage:
It seems that the number of motion detection alerts is being restricted somehow. Small movements are detected and reported correctly. If the movement ends, it will take a long time until the motion stop message appears.
In the meantime, new movements follow for which no new notification is sent.
I use motion detection for baby sleep tracking (e.g. movements per minute).

Long story short - my wish list for the Home Assistant Integration of Yi-Hack:
- MQTT motion detection more responsive
- MQTT-Auto Discovery
- Motion Snapshots as own camera 
- MQTT switches for the camera options

roleoroleo, 2020-07-31:
> It seems that the number of motion detection alerts is being restricted somehow. Small movements are detected and reported correctly. If the movement ends, it will take a long time until the motion [...]

Sorry but I can't change this behavior.
It's embedded on yi processes.
The hack only sniffs the ipc messages between processes and captures motion detections.

> Long story short - my wish list for the Home Assistant Integration of Yi-Hack:
> - MQTT motion detection more responsive
> - MQTT-Auto Discovery
> - Motion Snapshots as own camera 
> - MQTT switches for the camera options

Check the wiki, you can find examples about ha integration.
You can have snapshots and sensors.
Auto discovery and switches are not present and probably are a lot of code.

Pe-MaKer, 2020-07-31:
> Sorry but I can't change this behavior.
> Auto discovery and switches are not present and probably are a lot of code.

Thanks for that clarification.
It is as it is.

Any chance of getting rid of the "yi" watermark? I've seen that the yi-hack-MStar removed the original watermark image.

roleoroleo, 2020-08-01:
> Any chance of getting rid of the "yi" watermark? I've seen that the yi-hack-MStar removed the original watermark image.
> 
> 

See thus issue:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/34