roleoroleo/yi-hack-Allwinner #11: Online firmware upgrade loop

MisteurJ, 2020-04-26:
Hi,
thanks for this awesome work !
Tried to upgrade the fw online (0.1.3 to 0.1.4) but, after blinking orange, then blue, it's start again from orange to blue... Indefinitily !
I removed the SD card while it was blinking blue, and re-installed 0.1.3 on the SD.
All works so far, but I'm afraid to test again...

roleoroleo, 2020-04-26:
If you remove the sd the loop stops (I think...).

Download again the release and try to upgrade.
I uploaded temporarily the wrong file.

Or try again the online update.

MisteurJ, 2020-04-26:
Strange, I'll test later to upgrade to 1.0.5 :-)

roleoroleo, 2020-04-26:
If you want to try, edit /home/yi-hack/version to a lower number.
And then try upgrade online.

MisteurJ, 2020-04-27:
Do not want to try now, all is working...
When will you release the next version ? ;-)
By the way, I tried the ONVIF server, but I can't use it with my Synology NAS... Only the RTSP server works...

roleoroleo, 2020-04-27:
Please, check if it works with onvif device manager.

MisteurJ, 2020-04-28:
Yes, it works with ODM !

MisteurJ, 2020-05-10:
Hi,
tried to upgrade online to 0.1.5 today , and it works ! No reset loop anymore, when upgrade is done, cam is on and http server ready and reachable. Factory.done folder created on the SD card !
:-)

closed 2020-05-10 by MisteurJ