roleoroleo/yi-hack-Allwinner #13: the camera reboot when i request a snapshot

davimas, 2020-04-27:
Hey,

http Webinterface under 8080 works, but when i go to snapshot - the camera reboot aggain...

i have flashed the new 0.1.4 and try it with the older 0.1.3 version - same Problem.
I take a another new Camera (came 4 today) and the same issue - first snapshot is ok, when i reload the Snapshot in the Webinterface, or on the _http://IP-CAM:8080/cgi-bin/snapshot.sh_ the camera reboot..... :-/

(and that is no change if the SD Card is inside or outside)

Firmware Version | 0.1.3
Base Version | 8.2.0.0A_201912270941
Model Suffix | y20gam
Serial Number | BFUSY31IDQBAT....................
Local Time | Mon Apr 27 15:49:52 GMT 2020
Uptime | 0h 6m 19s
Load Average | 1.80 1.24 0.56
Free/Total Memory | 25772/61164 KB
Free SD Space | 99%

by, David

roleoroleo, 2020-04-27:
See here:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/5

davimas, 2020-04-28:
Okay.....this is not so good. I need connect the cams to Motioneye, but the rtsp stream is terrible, only snapshot has a good quality. Dammit - only 18 Euro from Amazon.de > great price :-/

I send the cams today back and order the xiaofang Cam version.

Thanks, by

closed 2020-05-04 by crdnl