roleoroleo/yi-hack-Allwinner #14: Notification on noise

KorbinianP, 2020-04-29:
Hello,
I received two of the Alwinner versions of the Yi 1080p Home Camera from amazon.de today. The first one I already flashed with v0.1.4, working nicely.

I wonder if I can get a notification based on audio? Maybe via mqtt? Then it would be very usefull as babycam. My kid moves quite a lot during sleep, motion detecion will either give loads of false positives (if too sensitive) or will not trigger at all (if not sensitive enough). On the other hand audio level should be quite easy.

Greetings

roleoroleo, 2020-04-30:
If I am not mistaken, this model is able to recognize baby crying.
I can probably intercept the ipc message (if it exists).
But I don't think it can be done with any noise, only crying. I can't control microphone and/or AI.

KorbinianP, 2020-04-30:
That sounds great! For my use case not any noise is required but baby crying. Only testing will show in the end, but sounds like a perfect fit!

roleoroleo, 2020-05-01:
I captured the ipc command for baby crying.
I think I can add to the mqtt messages.

roleoroleo, 2020-05-02:
Added support for baby crying.
Wait for the next release.

closed 2020-05-22 by crdnl