roleoroleo/yi-hack-Allwinner #17: RTSP audio and watermark

JakobStadlhuber, 2020-04-30:
Is there a way to get audio and a timestamp watermark?

roleoroleo, 2020-04-30:
Timestamp only in the snapshot.
Audio if someone is able to extract it from alsa.

JakobStadlhuber, 2020-04-30:
Would be nice future features if you need some ideas :)

Keep up the good work and thanks you for this project.

roleoroleo, 2020-05-06:
I copied the @PieVo idea and patched alsalib to duplicate the audio stream on a temporary pipe.
This way I can send the audio stream to the RTSP server.
I am testing the solution because I have encountered the same problems described here:
https://github.com/PieVo/yi-hack-MStar/issues/3

KorbinianP, 2020-05-11:
Do I understand correctly that I need to decide for the moment if I want Auto or Video? Not both at the same time? Will this change in the future?

roleoroleo, 2020-05-11:
You can choose video and audio or only video.
Not only audio.

KorbinianP, 2020-05-11:
OK, tnx for the clarification. After another restart of the camera the Video and Audio both showed up.

closed 2020-07-14 by roleoroleo