roleoroleo/yi-hack-Allwinner #18: YI outdoor support

JakobStadlhuber, 2020-04-30:
Do you have plans to support the YI outdoor cameras?

roleoroleo, 2020-04-30:
If based on the same platform  why not.
To start I need a log of the boot.

JakobStadlhuber, 2020-04-30:
> If based on the same platform why not.
> To start I need a log of the boot.

At the moment i do not have an Yi outdoor because your projects don't support it. Maybe someone else can help.

JakobStadlhuber, 2020-04-30:
You also implemented the feature for the yi dome to pan the camera, but only the Yi 1080p Home are listed as supported cameras, which haven't a pan feature, is it possible to flash a YI dome camera?

roleoroleo, 2020-04-30:
No.
This fw derives from MStar which is availabe for dome models too.
I didn't remove the code.

bytesex, 2020-05-03:
Hi! I have a "Yi Outdoor Camera 1080P" model YHS.3017
Current firmware version 3.0.0.0D_201809111054
Looks very solid and sturdy. I was going to make a dump from it, but it looks non-separable.
Are there chances of success if try to upload the firmware on it?

[image]
[image]
[image]

roleoroleo, 2020-05-03:
Before to try the hack we need to know which platform is.

bytesex, 2020-05-03:
It seems https://github.com/TheCrypt0/yi-hack-v4 worked for this :)
it seems thats a Hi3518e

JakobStadlhuber, 2020-05-03:
> well https://github.com/TheCrypt0/yi-hack-v4 worked for this :)
> it seems thats a Hi3518e

But you will have to pay/support for RTSP support. And the RTSP implementation is also not open source.

crdnl, 2020-05-04:
Considering the fact that it's covered by the yi-hack-v4 project, it's non applicable to this. If you want the license for that other project, go donate.

closed 2020-05-04 by crdnl

matteopiccioni, 2020-06-24:
The main problem with TheCritpt0 firmware is the miss mqtt and webservices support
I hope you will support YI outdoor too :)

roleoroleo, 2020-06-24:
The problem is not indoor or outdoor but the platform.

Hisilicon?
MStar?
Allwinner?

UlfgarIlFabbro, 2021-02-06:
> 
> 
> The problem is not indoor or outdoor but the platform.
> 
> Hisilicon?
> MStar?
> Allwinner?

how do i see which platform the cam is using