roleoroleo/yi-hack-Allwinner #21: RTSP performance

alexmc1510, 2020-05-05:
After configuration, enabling only ONVIF and RTSP (both with low Q) I loose camera connection randomly and camera reboots also from time to time. It seems a lack of performance. Are you aware about it? Do your know what can be done to improve it?

roleoroleo, 2020-05-05:
Cloud is enabled? Because cloud wastes a lot of memory.
If cloud is disabled and you choose only low quality, it's not a memory/cpu problem.
There is probably another problem.
Wait for the next release, I made a lot of changes on RTSP.

alexmc1510, 2020-05-05:
Yes, cloud is disabled.

crdnl, 2020-05-05:
@alexmc1510 

Can you provide some additional context on this issue:
- Camera Model
- yi-hack Version:
- Camera Version:

Can you export your settings from the web UI or give a quick overview of any changes that you've made?

crdnl, 2020-05-22:
Closing without response from issue filer for >14 days.

closed 2020-05-22 by crdnl