roleoroleo/yi-hack-Allwinner #24: Web UI does not properly save and show current settings

kevinhaendel, 2020-05-09:
Hello,
I just upgraded the firmware to `0.1.5`, and since this version, the Web UI does not seem to show the actual current configuration (under `/home/yi-hack/etc/system.conf`) and also doesn't seem to save the settings when clicking on the save button.
But this bug only applies to the Configuration Tab inside the Web UI (as far as I can tell (!)).

JakobStadlhuber, 2020-05-09:
Have the same issue, also upgraded via Online FW upgrade. I can succefully edited the files via ssh but it does not show up properly in the Web UI.

roleoroleo, 2020-05-10:
Do you use special chars in the config (for example the password)?
I changed the format of the set_configs.sh and I may have introduced a bug.
Try to overwrite the system.conf file with the default and try to reconfigure the cam:

```
HTTPD=yes
TELNETD=yes
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=no
REC_WITHOUT_CLOUD=no
MQTT=no
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=none
ONVIF=yes
ONVIF_PROFILE=high
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
FREE_SPACE=0
TIMEZONE=
```

roleoroleo, 2020-05-10:
Or post your config (if you can)...

det-box, 2020-05-10:
Hi,
Have the same issue, also upgraded via Online FW upgrade.
Time is 2 hours behind in Europe (CET-1CEST).
The TimeZone and the hostname are also not saved.
Where can I edit them?

Thanks

roleoroleo, 2020-05-10:
Login with ssh and edit with vi.

roleoroleo, 2020-05-10:
I tried to downgrade/upgrade again and it worked.
I can't replicate the bug.

JakobStadlhuber, 2020-05-10:
> Do you use special chars in the config (for example the password)?
> I changed the format of the set_configs.sh and I may have introduced a bug.
> Try to overwrite the system.conf file with the default and try to reconfigure the cam:
> 
> ```
> HTTPD=yes
> TELNETD=yes
> SSHD=yes
> FTPD=yes
> BUSYBOX_FTPD=no
> DISABLE_CLOUD=no
> REC_WITHOUT_CLOUD=no
> MQTT=no
> RTSP=yes
> RTSP_STREAM=high
> RTSP_AUDIO=none
> ONVIF=yes
> ONVIF_PROFILE=high
> ONVIF_WM_SNAPSHOT=yes
> NTPD=yes
> NTP_SERVER=pool.ntp.org
> PROXYCHAINSNG=no
> RTSP_PORT=554
> ONVIF_PORT=80
> HTTPD_PORT=8080
> USERNAME=
> PASSWORD=
> TIMEZONE=
> FREE_SPACE=0
> ```

I reseted the config and eddied it afterward in the UI, now its working fine.

JakobStadlhuber, 2020-05-10:
Previous settings: https://hastebin.com/teqeqamomu.ini

det-box, 2020-05-10:
There is no entry in system.conf for the hostname.

roleoroleo, 2020-05-10:
/home/yi-hack/etc/hostname

det-box, 2020-05-10:
Thanks,
The file hostname is not created for the Web interface save. If a new file was created with the vi, the host name is iO.
When I click on save in the web interface, the system.conf file says
DISABLE_CLOUD=&PASSWORD

created  with vi
`root@Cam2:~# cat /home/yi-hack/etc/system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=no
REC_WITHOUT_CLOUD=yes
MQTT=yes
RTSP=yes
RTSP_STREAM=low
RTSP_AUDIO=low
ONVIF=yes
ONVIF_PROFILE=both
ONVIF_WM_SNAPSHOT=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
FREE_SPACE=0

make the backup in the web interface root@Cam2:~# cat /home/yi-hack/etc/system.conf
`root@Cam2:~# cat /home/yi-hack/etc/system.conf
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=&PASSWORD
REC_WITHOUT_CLOUD=yes
MQTT=yes
RTSP=yes
RTSP_STREAM=low
RTSP_AUDIO=low
ONVIF=yes
ONVIF_PROFILE=both
ONVIF_WM_SNAPSHOT=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
FREE_SPACE=0

The time zone of the system is not correct

roleoroleo, 2020-05-10:
Did you try to clean the cache of the browser?

det-box, 2020-05-10:
Thanks,
After deleting the browser cache it works now.

closed 2020-07-14 by roleoroleo