roleoroleo/yi-hack-Allwinner #28: BFUSY31 cam impossible to hack

fredmassa, 2020-05-11:
Hi Roleo,
I have a Yi home 1080p with BFUSY31... with firmware 8.2.0.0A_201912270941.
I downloaded the y20gam_0.1.5.tgz file from https://github.com/roleoroleo/yi-hack-Allwinner/releases, formatted the SD using Fat32 (I tried with a 64Gb SD first and then with a 16Gb SD). I put the y20gam_0.1.5.tar file (what i have inside the archive) into the SD and then I turn on the cam but nothing happens.

The problem is that the cam start as normal withou flashing the firmware. The yellow led start to flash but after that became blu. If I reset the cam before to put the SD, when it turn on starts speaking "Waiting to connect" .

Can you help me?

Many thanks

digitalchou, 2020-05-11:
you have to decompress the tar and put the two folders and one file on the rrot of the sd card. But i have the same exact model than you and after flashing i cant conect to the cam, see the issue I opened 3 minutes later. Im blocked, wait for anyone that can help us

fredmassa, 2020-05-11:
@digitalchou can you link your issue?

digitalchou, 2020-05-11:
#29 this is my issue

fredmassa, 2020-05-12:
> #29 this is my issue

Thanks

fredmassa, 2020-05-12:
ok, i formatted again the SD and I moved only the 2 folders with the home_y20gam.stage file. I moved the Sd to the cam and turned it on. This time something happen... yellow flashing light, yellow fix light, "click", again yellow flashing light, yellow fix light and then "waiting to connect" message. I made the pairing with the app and everything work.

Now I try to telnet the cam but after the login (only root) I've this problem "_mounting none on /sys/kernel/debug failed: Resource busy_ ". 
[image]

I tried to change the password but I've this error " _can't update password file /etc/passwd_ "
[image]

roleoroleo, 2020-05-12:
Don't worry.
/sys/kernel/debug error is normal
You can't change the root password, the root file system is read only.

fredmassa, 2020-05-12:
ahn! ok 👍 
so now i can see the configuration here http://CAM_IP:8080/ and .... the hack was successful!!!
[image]

att1x, 2020-05-12:
Hi,

I got the exact same problem, also BFUSY31. The firmware just does not flash. The internal format function does not remove e.g. the "record" folder, is this normal? Neither way of formatting it get's the device to flash the firmware though. Very sad.

roleoroleo, 2020-05-12:
Try to change the SD card.
Try to format it from a PC.

Gerd2000, 2020-05-15:
Hello,
i have also a new cam with serial no BFUSY31321. I have nothing installed before and also no connection to the internet. 
The hack from here (Allwinner) was not loadet.
Can i start direct with the firmware hack or must i setup my cam at first like the manufacturer said? I have no fast internet connection and would not start with this setup and also not install any app at my mobile.
Thank you,
Gerd

crdnl, 2020-05-22:
@Gerd2000 

You must setup the camera using the manufacturer directions to get the camera connected to the Wireless Network (Wi-Fi). After the camera has connected to the network, you can proceed to install the modded firmware, and disable the cloud features from within the Web UI.

ilmisu, 2020-06-24:
> ok, i formatted again the SD and I moved only the 2 folders with the home_y20gam.stage file. I moved the Sd to the cam and turned it on. This time something happen... yellow flashing light, yellow [...]

Please help me as well... I cannot flash my BFUS camera at all... :( same stock fw.. I don't know if maybe is the micro SD card... I tried to format it from the camera and then add just the folder, I formatted it from laptop, but nothing happens. I use a 16gb Sandisk microsd...

Merlynda, 2020-07-01:
Camhash?
Yi home 1080p BFUSY31. Yi hack Allwinner worked. However, there is no RTSP tab. I have the RTSP server radio button enabled. 

How do I find the camhash so I can donate and stream RTSP? Thx

roleoroleo, 2020-07-01:
You don't need to donate.
rtsp implementation is open source.
Simply try and use it.

But if you want to donate follow the link at the bottom of the main page. :)

roleoroleo, 2020-07-14:
Sorry for delayed response.
You don't nedd to donate to use RTSP.
Enable it in the Configuration tab.

closed 2020-08-29 by roleoroleo