roleoroleo/yi-hack-Allwinner #29: Flashing is very short, cant conect anymore

digitalchou, 2020-05-11:
I flash mi yi 1080p home BFUSY31CG1 with 8.0.0XX firmware.
I put the SD card with the 2 folders and one file , then I power on the cam.
Blue led flashes and i heard thye relay "click", then blue led flashes a couple of seconds and turns yellow. after a few seconds, the "click" again, the cam reboots flashing blu led a few seconds and then it conect to network, but i cant acces anymore. I read the wiki webcodes and noone work.
I saw in the network but i cant conect any way,
The flasin proces is incomplete im afraid.
I tray version 1.4 and the same thing happens.
I have return to flash last version, , 1.5 , with identical result,I cant conect anymore. 
I have another identical cam and saw in the wiki the process to bacup stock firmware, but im dont know how to flas it again on the cam that dont work.
But id like to have rtsp streamin to my computer, so it will be wonderfull if this firmware works for me , any ideas?

Sorry for mi english

digitalchou, 2020-05-11:
well, after exploring a little I find that the cam conects to mi network, Ican reset and reconect but it dont pair with the android app, I cant acces anyway.
I think config.sh on the sd card its executed, it disables yi junk in init.sh , and repalces the update.sh wtih a more friendly one, but then for some reason the firmwareis not flased, it reboots with the old firmware but with almos services disabled.

This is a backup of the system after flashin with yi-hack-Allwinner 0.1.5,
the current state: 
[backup_POST FLASH.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/4612509/backup_POST.FLASH.zip)

And this is a backup from the other cam in the pack, as it comes from factory
without any modification, conected to my network and working flawessly...
but witout rtsp:
[backupFACTORY+WIFI.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/4612510/backupFACTORY%2BWIFI.zip)

Firmware version is: 8.2.0.0A_201912270941
and the exact model: 1080p home BFUSY31CG1

Hope it helps, thank you for your work

pbanj, 2020-05-12:
have you made sure nothing on your network is using port 8080?

roleoroleo, 2020-05-12:
Looking at the backups you've made, there are a lot of strange things.
There are traces of another hack, probably yi-hack-v3.
Original init.sh, update.sh and extpkg.sh are missing.
It takes a lot of work to recover your cam.

digitalchou, 2020-05-12:
Many thanks, roleoroleo.
Yes, I find this hack at the end of a large seek, and I try another hack before, but it seems not do anyting, the cam start like ever and I though the sd card update was not working,
there was no changes that I could see.
The lack of extpkg.ssh may explain that I cant access trhu ssh, ill try with telnet.
When I have time I´ll study it a bit and I´ll try to recover.

I have another identical cam, lookin at the "backupFACTORY+WIFI.zip" (extracted from this second cam):
Will the hack work? That cam is as out of the box, the only thing I do with it is conect to my network via the android app,I made testing only in the first cam.
It will be nice if it works, i can sacrifice one of the cams, its a cheap toy,
but I want rtsp for streamin, 
Im doubt about the firmware version, thus its a brand new cam bougth a couple of days ago.
I read users with the same model BFUSY31CG1 working, but i´m not shure if mi firmware is the same version or a newer one that isnt supported.

Many thanks for your time.

roleoroleo, 2020-05-12:
The hack should work if the fw is 8.2.0.0A
Check the fw version on your app.

I will try to create a script to recover your 1st cam.

digitalchou, 2020-05-12:
Many thanks, roleoroleo.

The app says that firmware version is: 8.2.0.0A_201912270941
It seems that 201912270941 is a timestamp of the compilation,
so I will try formating sd card , decompress y20gam_0.1.5.tgz and paste the contents on the root of the sd card. Then I put the card on the cam and power on. 
The led would blink on yellow a while, the cam would reboot and the led would blink yellow again for a long time (till an our. as I read on the guide on the main page of the proyect). The process is finished when the led turns blue and dont blink anymore.

Im rigth?

I have two more questions:
I need to reboot the cam at last to it boot to the new firmware, or I can use it when the process finish?
Should I reset the cam before doing this, or leave it as is, already conected to my network?

It would be nice if you do that script to recover the first cam, I used linux years ago, and I was unwrapped with the terminal, but I never understand scriptin on linux very well.
Without ssh way(ssh its not working), I´m trapped.

Thank you so much for your effort and your patience!

roleoroleo, 2020-05-12:
Correct.
If you use a clean SD card the upgrade procedure will finish in a minute.

The cam will reboot at the end of the procedure. When the system restart is ready, no more reboots.

Don't reset the cam before the upgrade.

roleoroleo, 2020-05-12:
Try this for your bricked cam:
https://drive.google.com/open?id=1Jp0rQvP1WqkPQDEl34PlK4pZsJVJK18l

Clean the sd card before try.

digitalchou, 2020-05-12:
Thanks roleoroleo , it works perfectly, after your repair the cam conect to network via android app perfectly, paired and working perfectly. 
Now I go to try y20gam_0.1.5 , and post here the results.
Im crossing my fingers...
Thank you very much

digitalchou, 2020-05-12:
well, this is working fine, mi two cams are now flased and work correctly.
MAny thanks for you effort roleoroleo.
I cant see a button to contributeon your proyect, or in your profile, is there anyway to donate?

The rtsp performance is not good even with all other services turned off, and OBS hangs.
On VLC it runs better, but i have to investigate how to midify encoder parameters.
It be nice to reduce the high rtsp to 1280x720 to get better performance,thus low its very low, i will put that in pull requests better.

Many thanks again for your effort and your patiente.
I wait a while to close the thread for the donate question.

pbanj, 2020-05-12:
His donation info is at the bottom of the readme/main page of the github
https://cdn.discordapp.com/attachments/399670410565910529/709811916301664799/Screenshot_20200512-125757.png

digitalchou, 2020-05-12:
thanks , I close this thread

closed 2020-05-12 by digitalchou

roleoroleo, 2020-05-12:
> The rtsp performance is not good even with all other services turned off, and OBS hangs.
> On VLC it runs better, but i have to investigate how to midify encoder parameters.
> It be nice to reduce the [...]

It's not possible to change resolution: you can choose only between 1920x1080 and 640x360.
Unfortunately, we don't have control on the encoding phase.