roleoroleo/yi-hack-Allwinner #30: Additional unix tools

Xandrios, 2020-05-12:
Hi Roleo, thanks for your work on this project. Once I have my cameras working I'll be sure to make a donation for your efforts.

I've just received a set of the Yi-Home cams and they turn out to be the BFUSY31 type. Successfully backed up the original firmware using the backup-script-on-SD-card method and flashed to the 0.1.5 hack version. Seems to work well so far.

There are a few things I ran into and I was wondering on your opinion.

The MQTT doesn't seem to generate any events, likely because of [this](https://github.com/roleoroleo/yi-hack-Allwinner/issues/25#issue-615471614) existing issue. 

My goal is to implement a 'poor mans cloud' where the device will upload movement to an offsite location, or alternatively, have an offsite script download movement videos when MQTT events come in. The absolute best way to have this platform working would be to have it continuously record to SD card 24/7, and send movement videos off-site... but I'm not sure if that is possible (There are no separate movement-videos when 24/7 recording is enabled right?). 

In order to just sync the record directory to an external location we'll need some tools, most suited would be rsync. Though I noticed that rsync is not currently available on the platform. Is there any way to add these to either /home/yi-hack/bin or by perhaps utilizing the space available on the SD card (and adding that bin folder to the PATH)? 

Lastly I wasn't able to set the root password and I noticed / (including /etc) to be ro-mounted. Available space on that is 0 bytes as well, which explains why this is the case. Do you have any suggestions as to securing the camera - especially if we want to make services like sftp publicly available to have a remote script fetch recordings?

Your work on this project is much appreciated!

roleoroleo, 2020-05-12:
> The MQTT doesn't seem to generate any events, likely because of [this](https://github.com/roleoroleo/yi-hack-Allwinner/issues/25#issue-615471614) existing issue.

Yes, wait for the next release.

> My goal is to implement a 'poor mans cloud' where the device will upload movement to an offsite location, or alternatively, have an offsite script download movement videos when MQTT events come in. [...]

Correct, there are no saparate videos for 24/7 and motion detection.
But you can download video files from motion start event to motion stop event.
See the wiki about web services.

> In order to just sync the record directory to an external location we'll need some tools, most suited would be rsync. Though I noticed that rsync is not currently available on the platform. Is there [...]

Probably I can add to the next release.
But take a look to this: https://github.com/roleoroleo/yi-hack-MStar/issues/77

> Lastly I wasn't able to set the root password and I noticed / (including /etc) to be ro-mounted. Available space on that is 0 bytes as well, which explains why this is the case. Do you have any [...]

You should use a service for which you can configure a minimal security, for example http.

Xandrios, 2020-05-12:
Thanks, version 0.1.6 resolves the MQTT issue and general SWAP/memory problems as well. Very nice.

I've verified that MQTT events are still coming through even if 24/7 recording is on. So those can indeed be used with the web services using username/password authentication. 

Additionally it is very easy to extend the web services with bash scripts hosted on the sd card (/www/cgi-bin) if the following symlink is made after flashing:

`ln -s /tmp/sd/www/cgi-bin/ /home/yi-hack/www/cgi-bin/sd
` 
After which those custom scripts can be executed by calling http://<ip>:<port>/cgi-bin/sd/<script>.sh. This may prove to be useful.

My preference though would be to push videos to a remote source, like the Gdrive script that you linked. Because that resolves the problem of having to assign static IP's to the camera's and create portmappings through the NAT router that the cameras will be behind.

Gdrive is not my personal favorite though, I'd rather push the data using sftp or rsync to a remote NAS or linux server as it is less complex, and more flexible as these are more standard protocols. I'm not sure how difficult it would be to port sftp/rsync though - especially as we are then talking about SSL encryption.

roleoroleo, 2020-05-12:
I built rsync without problems.
But it's > 400 KB. A lot of space...

[rsync.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4618321/rsync.tar.gz)

Xandrios, 2020-05-12:
Thats awesome, thanks so much! Yes, it is large (relatively) ... but at the same time we have SD cards of GBytes in size, right? Is there a reason to not host the binaries and scripts on the SD card permanently?

If I put that rsync executable on the SD card it works without a hitch. In the yi-hack home dir I even found an SSH binary that I was not aware of (Its not on Path by default), and combined that allows to securely transfer files using rsync/ssh:

`/tmp/sd/rsync -avzhe '/home/yi-hack/bin/ssh -p <port>' /tmp/sd/test_input_dir/ <user>@<server>:/<path>/`

This will synchronize all changes between input and dir and destination, via rsync-over-ssh. The beauty of this is that it only synchronizes the changes, not all data. Theoretically one could even synchronize the recording-temp-file while a recording is going on, as to make sure that as much video material has reached the remote server before the camera is (potentially) disconnected.

roleoroleo, 2020-05-13:
> Thats awesome, thanks so much! Yes, it is large (relatively) ... but at the same time we have SD cards of GBytes in size, right? Is there a reason to not host the binaries and scripts on the SD card permanently?

I don't want that the fw to depend on the presence of an sd card.

> If I put that rsync executable on the SD card it works without a hitch. In the yi-hack home dir I even found an SSH binary that I was not aware of (Its not on Path by default), and combined that [...]

It's not in PATH for the root shell but is in PATH for system.sh execution.
This problem is always due to the fact that rootfs is read only.

> Would there be a way to, in a script on the cam, get motion/movement triggers similar to MQTT? Do we know where the MQTT service gets its triggers from?

There is a message queue. See mqttv4 sources.

Xandrios, 2020-05-13:
Thanks. Found the POSIX /dev/mqueue/ipc_dispatch queue for messages - unfortunately I don't have a way to read that queue without compiling an app for that (using mqueue libs). I'm not currently setup to cross compile.

Is it the native Yi software that generates these events? Looking at your recent baby-crying commit it does seem that these events are being generated for us, put on the queue, and we should just be able to read them. Unfortunately we don't have any unix binaries to read queues...not sure of those exist.

Maybe I'll try to get crosscompile to work.. but that will require some additional time to figure that out. This is not anything I'm used to working with :)

pbanj, 2020-05-13:
Maybe instead of having the firmware rely on an SD card you can have it be an option or have a folder that it can check on boot and load stuff from. So if it isn't there it's ignored and everything carries on like it is now.

roleoroleo, 2020-05-13:
> Thanks. Found the POSIX /dev/mqueue/ipc_dispatch queue for messages - unfortunately I don't have a way to read that queue without compiling an app for that (using mqueue libs to allow to just utilize mq_receive). I'm not currently setup to cross compile.

The cross-compile environment is alredy compiled. See the wiki.

> Is it the native Yi software that generates these events? Because in that case hooking in to these events may be difficult. Looking at your recent baby-crying commit it does seem that these events are being generated for us, put on the queue, and we should just be able to read them. Unfortunately we don't have any unix binaries to read queues...not sure of those exist.

Message queue is the technology that Yi has chosen as ipc communication.
The main program (rmm) reads the stream, processes it and generates the ipc messages for all other programs.
We dont'have a unix binary to read mq but we can build it.

> Would you think that adding a reader (similar to mqtt) that simply calls different scripts upon the various events be useful? It'd allow easy development of 3rd party/custom functionality based on motion/crying events.

I don't know if it could be useful. It depends on the needs.

Xandrios, 2020-05-14:
> The cross-compile environment is already compiled. See the wiki.

Thanks, yes. On what platform do you run the compilation? I've installed a Redhat 7 VM, added the toolchain and installed a whole bunch of dependencies, but it fails on a compile flag seemingly related to yacc. My impression is that the compile script expects to run with a different compiler? I'm not too familiar with these things; but I imagine it may help if I match your build OS/version.

`./scripts/compile.sh`

> make[5]: Entering directory `/home/siet/yi/yi-hack-Allwinner/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap/src'
gcc -DWITH_YACC -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/im                                                                 port\"" -DLINUX -g -O2   -o soapcpp2 soapcpp2-soapcpp2_yacc.o soapcpp2-soapcpp2_lex.o soapcpp2-symbol2.o soapcpp2-error2.o soapcpp2-init2.o soapcpp2-soapcpp2.o -ly
/bin/ld: cannot find -ly
collect2: error: ld returned 1 exit status
make[5]: *** [soapcpp2] Error 1

Xandrios, 2020-05-14:
> The cross-compile environment is already compiled. See the wiki.

Could I ask you on which distribution/architecture/version you compile this project? I've tried on a redhat 7 VM but (after installing a bunch of dependencies) it seems to fail on a compiler option related to yacc/flex:

> make[5]: Entering directory `/home/siet/yi/yi-hack-Allwinner/src/onvif_srvd/onvif_srvd/gsoap-2.8/gsoap/src'
gcc -DWITH_YACC -DWITH_FLEX  -DSOAPCPP2_IMPORT_PATH="\"/usr/local/share/gsoap/import\"" -DLINUX -g -O2   -o soapcpp2 soapcpp2-soapcpp2_yacc.o soapcpp2-soapcpp2_lex.o soapcpp2-symbol2.o soapcpp2-error2.o soapcpp2-init2.o soapcpp2-soapcpp2.o -ly
/bin/ld: cannot find -ly
collect2: error: ld returned 1 exit status

The dir /usr/local/share/gsoap/import didn't exist, after installing gsoap /usr/share/gsoap/import was created, which I symlinked as a desperate measure...but no go. I'm thinking this may be related to the compiler being different from yours?

>  gcc --version
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-39)

roleoroleo, 2020-05-14:
Maybe bison is missing.
I'm using debian.

Xandrios, 2020-05-14:
Thanks. It wasn't bison (Already had that installed), but it seems the compile scripts seem to have a problem with Redhat architecture. I've fired up a Debian VM and that works. 

I recommend using one of the cloud VM providers that allow you to quickly jumpstart VM's with a certain OS for a few cents per hour. I've used the following:

OS: Debian 10
Memory: 4GB (2Gb gives OutOfMemory during compile)
CPU: Dedicated (Choose whatever you want, but know that compiling is CPU-intensive and takes long)
Disk: SSD

Most of my time was spent on figuring out dependencies that had to be installed, as the compilation kept quitting for missing libraries (requiring to start all over again). 

The following steps are needed for installing the dependencies and a successful compile. I don't believe that I can update the Wiki, but feel free to copy some of this to allow others to easily start with building the firmware (and tools) themselves. 

```
## Get dependencies
apt-get install git unzip build-essential bison++ bisonc++ libbison-dev autoconf autotools-dev automake libssl-dev zlibc zlib1g-dev libzzip-dev flex libfl-dev yui-compressor closure-compiler optipng jpegoptim libtidy5deb1 node-less sassc sass-spec libhtml-tidy-perl libxml2-utils

## Get toolchain
git clone https://github.com/lindenis-org/lindenis-v536-prebuilt
mkdir -p /opt/yi/toolchain-sunxi-musl
cp -r lindenis-v536-prebuilt/gcc/linux-x86/arm/toolchain-sunxi-musl/toolchain /opt/yi/toolchain-sunxi-musl/

## Get yi-hack
git clone https://github.com/roleoroleo/yi-hack-Allwinner
cd yi-hack-Allwinner/
git submodule update --init

## Compile
./scripts/compile.sh
## Package
./scripts/pack_fw.all.sh
```
I've compiled a v0.1.60 version this way as a test, and successfully flashed the camera. Haven't made any changes to the code yet though - thats for another day.

[image]

roleoroleo, 2020-05-14:
Thanks for your feedback.
I will update the wiki with your details.

Xandrios, 2020-05-15:
Thanks.

Thinking from a user perspective what I'd like to know is not only when movement/sound was detected, but what I REALLY want to know is...from all my video files, which events where observed during which videos. I'd like to see the following for example:

/tmp/sd/record/2020Y05M15D23H/11M21S38.mp4
/tmp/sd/record/2020Y05M15D23H/12M23S12.mp4
/tmp/sd/record/2020Y05M15D23H/13M48S33_MOVEMENT.mp4
/tmp/sd/record/2020Y05M15D23H/14M06S29_MOVEMENT_SOUND.mp4
/tmp/sd/record/2020Y05M15D23H/15M34S08.mp4
/tmp/sd/record/2020Y05M15D23H/16M03S54_CRYING.mp4
/tmp/sd/record/2020Y05M15D23H/17M14S07.mp4
/tmp/sd/record/2020Y05M15D23H/18M03S54_SOUND.mp4

Because this allows me to do something with these videos depending on what event was observed during the recording of the video. For example I could rsync all videos that detected movement, while leaving all other videos on the camera untouched.

As the video files are written by a Yi application we cannot define the filename. But we can modify the filename based on the events that came in while the video was being recorded. 

I see two ways of going about that:

1. Make a new module/binary that subscribes on the IPC POSIX events, monitors which files are being recorded, and renames files after they have been recorded based on the observed events. 
2. Allow external scripts to receive IPC POSIX events, and implement this functionality in a bash-like script. 

Personally I'd much prefer option #2 as I myself am way more confident with Bash than C. One way to do this would be to extend the ipc_cmd tool with support for receiving events (option '-e Receive events'?). When ipc_cmd is executed with that argument it subscribes on the queue, listens for messages, and once a movement/sound event comes in it stops reading from the queue, relays the message for other processes (e.g. mqtt) and exits with a certain exit code (e.g. 100 for movement start, 101 for movement end, 102 for sound, etc). 

This allows a script running in the background to call ipc_cmd to listen for events, read the exit code, rename recordings (or queue that if a tmp-file is being recorded to), and re-call ipc_cmd to be ready for the next event. 

This would also allow other scripts that don't necessarily care about the video files to receive movement/sound events easily.

What do you think about such approach?

Xandrios, 2020-05-16:
I've updated the ipc_cmd src. If you're interested I'll send a pull req. to get it merged:

> root@TestCam:~# while true; do ipc_cmd --event; echo $?; done
> MOTION START
> 100
> MOTION STOP
> 101
> MOTION START
> 100
> MOTION STOP
> 101

Can't get the baby crying to trigger, but the mechanism is the same as in the MQTT code.

roleoroleo, 2020-05-16:
mqttv4 already sends mqtt messages with start, stop and the list of mp4 files related to the event.
You already have all the informations you need to execute some job externally.

You could, for example, subscribe to the mqtt server and parse json messages (mosquitto_sub and libmosquitto are already compiled and available).
Or we could create a different output for the messages(fifo?), but I think it's not a good idea to exit mqttv4 and restart it. We could miss messages (for example, it often happens to have 2 start/stop events overlapping).

roleoroleo, 2020-05-16:
Another example...
I could clean stdout and write messages on it, so we can use a pipe with a shell script.

Xandrios, 2020-05-16:
> mqttv4 already sends mqtt messages with start, stop and the list of mp4 files related to the event.
> You already have all the informations you need to execute some job externally.

This is true, but relies on MQTT, so it needs an external MQTT broker to work. That adds delay and/or break functionality if the internet connection is unavailable.

> but I think it's not a good idea to exit mqttv4 and restart it.
Fully agree. But the ipc_cmd binary is specifically for this purpose right, to allow scripts to interact with the IPC queues. That's why I added it there, to basically extend the functionality of the ipc_cmd tool.

> I could clean stdout and write messages on it, so we can use a pipe with a shell script.
You mean the MQTT stdout? But that requires to use MQTT then, and it requires to connect/publish to an external server which you may not want if you are just scripting some local (on-camera) functionality?

roleoroleo, 2020-05-16:
> You mean the MQTT stdout? But that requires to use MQTT then, and it will have to connect/publish to an external server which you may not want if you are just scripting some local (on-camera) functionality?

No I mean ASCII output, something similar to what you wrote:
```
MOTION START 100
MOTION STOP 101
BABY CRYING 110
```

In this case, only one destination process.

> I think it would be important that either we can start the process multiple times (once for each reader process/script), or, have it start once and have multiple readers.

The current situation is already "critical" because mqttv4 removes the message from the queue to sniff it and then adds again to send the message to the original recipient.
If you start the process multiple times (reading and writing the same queue with the same time sequence) what happen? I don't know.

If you need multiple receivers we need to think another solution.
Maybe one ipc reader which share messages with multiple processes.

Xandrios, 2020-05-16:
Unfortunately you are correct. I tested running the the listener 3 times in parallel and generated a few events. The behavior is incorrect, though very repeatable:

1. Motion start events are always received by 2 out of 3 receivers
2. Motion stop events are always received by all 3 receivers

The receiver that is not getting a motion-start is not the same one each time. But, there is one receiver not getting the start event with every attempt.

Reducing the receivers to 2 does not resolve the problem. Just one of the two would be getting the start event. Both get the stop event. 

Reducing the receivers to 1, having MQTT off, works quite reliably it seems. This means that indeed the mechanism of 're-sending' the received events is not working correctly. It's odd though that stop-events are working as expected. I'd almost think that we have some more listeners that do not relay the start-message properly?

Conclusion is... that we would probably indeed have to move towards a single application that receives the IPC events. And that application would have to distribute the events to MQTT and other listeners using some other method. Question is, what is an easy to use method of distributing events from one source to multiple listeners, where the listeners can be C-based or a script (Bash)..

Xandrios, 2020-05-16:
Update: Found the cause. The Yi software. 

When disabling the Yi-Cloud function all events come in, to all listeners, as expected. Seems like the Yi-Cloud listener performs some kind of action upon the Motion-Start event, and does NOT relay the event back to the queue for other listeners. Disabling the Yi-Cloud resolves that.

So I guess that now the question is... would it still be worth overhauling the event handling, or now that we know that disabling the Yi-cloud (which is something that everyone should do anyway) resolves it.

roleoroleo, 2020-05-16:
Yes, mqttv4 is a hack that sniff messages between 2 yi programs.
Normally, the destination program receive the message and remove it from the queue.

And we can't disable cloud totally, a lot of users use yi app.

Xandrios, 2020-05-16:
So basically this means that IPC events, including MQTT, only works reliably if the Yi-Cloud features have been turned off. Even if we were to make a special application to receive IPC event and distribute them further to other listeners, it would not be 100% reliable as the Yi-software may still consume some events in some cases.

I think I'll go ahead with the current ipc_cmd change then, at least for now. This seems to work fine when the Yi-Cloud is turned off, I have three separate sessions read events concurrently for the past hour or so...everything is fine and in sync. They also match up with MQTT events.

roleoroleo, 2020-05-16:
If the receiver is only 1 the reliability is high enough.
I have never noticed loss of messages until now.

I'm working on a new version of mqttv4 that forwards messages in new queues (ipc_dispatch_1, ipc_dispatch_2, etc...).
And a client that reads messages and output them in stdout.
In this manner you can have 10 processes togheter.

roleoroleo, 2020-05-16:
Try this:
[queues.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4639072/queues.tar.gz)

Xandrios, 2020-05-16:
Thanks. Doesn't quite work yet as the queues have not been created:

root@TestCam:~# ipc_read -n 1
Can't open mqueue /ipc_dispatch_1. Error: No such file or directory
root@TestCam:~# ls -all /dev/mqueue/ipc_*
-rw-r--r--    1 root     root            80 May 16 20:44 /dev/mqueue/ipc_cloud
-rw-r--r--    1 root     root            80 May 16 20:47 /dev/mqueue/ipc_dispatch
-rw-r--r--    1 root     root            80 May 16 20:44 /dev/mqueue/ipc_dispatch_worker
-rw-r--r--    1 root     root            80 May 16 20:44 /dev/mqueue/ipc_p2p
-rw-r--r--    1 root     root            80 May 16 20:44 /dev/mqueue/ipc_rcd
-rw-r--r--    1 root     root            80 May 16 20:44 /dev/mqueue/ipc_rmm
-rw-r--r--    1 root     root            80 May 16 20:44 /dev/mqueue/ipc_rtmp

But, having this in MQTT, it requires a working MQTT connection. If the MQTT connection is not set up succesfully, the IPC stuff within MQTT4 is not loaded and there won't be any events. In my opinion MQTT may not be the best suited place for this general functionality? Can we move this to a standalone app and have MQTT use of those those 10 incoming channels?

roleoroleo, 2020-05-17:
Yes, we could make this change.

roleoroleo, 2020-05-17:
Try this:
[queues.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4640030/queues.tar.gz)

Start ipc_multiplexer before mqttv4 (if you use it).
mqttv4 uses /ipc_displatch_1 queue.
From 2 to 9 are free.

Xandrios, 2020-05-17:
Thanks, that seems to work well. Two details that I noticed that maybe we can look at, but overall this works. 

1. When doing a read on a queue not read before, you will get all previous messages that were published in the past. This means that if you have the camera running for a while, once you execute the 'ipc_read' command you may get about ~10 old/expired events first, before it starts streaming realtime data. It seems that the queue is somehow 'holding' these messages for a while if there is no reader. Is there a way to check if there is a reader on a queue, and if not, to not publish to that specific queue?

2. The ipc_read usage is, I think, not correct...it should only work for n/d/h arguments right?

> Usage: ipc_read [t ON/OFF] [-s SENS] [-l LED] [-v WHEN] [-i IR] [-r ROTATE] [-m MOVE] [-p NUM] [-d]

Maybe when adding this to the next release it may be good to mention in the documentation and/or --help of cmd_read to _only_ use queues 5-9 for custom scripting. This allows queue 1-4 to be used for 'official' features like MQTT, and additions that may be added later. That way new features in the hack-firmware will not conflict with scripts that people may make using these queues.

roleoroleo, 2020-05-18:
> Thanks, that seems to work well. Two details that I noticed that maybe we can look at, but overall this works.
> 
> First, small detail: The ipc_read usage is, I think, not correct...it should only work for n/d/h arguments right?
> 
> > Usage: ipc_read [t ON/OFF] [-s SENS] [-l LED] [-v WHEN] [-i IR] [-r ROTATE] [-m MOVE] [-p NUM] [-d]

Ok, done.

> When I start reading queue 8 I would be getting all those 92 messages at once, and only then it would start streaming 'realtime'. Also, I don't know what happens once a queue has thousands of events 'waiting'. Something may break or overflow at that point.
> 

The max size of the queue (when created) is 10, so we can have max 10 messages on the queue.
I could add a "clearing" function that clean the queue before start the receive (ipc_read).

> When adding this to the next release it may be good to mention in the documentation and/or --help of cmd_read to _only_ use queues 5-9 for custom scripting. This allows queue 1-4 to be used for [...]

Ok.

> Alternatively, maybe, would it be easier to just append the events to a file somewhere? Processes that want to be notified can just tail that file and read any event that comes in. It would not limit the number of receivers, and it would work even if there are 0 receivers. The only catch is, that the file would have to be cleaned up, without the tailing applications breaking.
> 
> Using the official 'tail' command that works. For example I have a sender and receiver:
> 
> ```
> # tail -f /tmp/sd/testfile &
> # echo "MOTION_START!!" >> /tmp/sd/testfile
> MOTION_START!!
> # echo "" > /tmp/sd/testfile 
> 
> # echo "MOTION_STOP!!" >> /tmp/sd/testfile
> MOTION_STOP!!
> ```
> 
> Even if I empty the testfile and then append more events, they show up fine in tail. Though I don't know if the same happens when you read/tail the file through a C application...that may not work as easily. You may have to reset reading from the beginning if the filesize gets smaller. Maybe just call the external tail-binary from within C? ;-)

In case, if we have problems with the queues...

Xandrios, 2020-05-18:
Sounds good.

closed 2020-08-29 by roleoroleo