roleoroleo/yi-hack-Allwinner #32: Turn on/off the cam with HA automation?

fredmassa, 2020-05-13:
Hi @roleoroleo ,
many thanks for everything! Your work is great!!
I 've a question about using the hacked Yi home 1080p camera with HA. Is it possible to turn on/off the cam with an automation in home assistant?

Thanks ;-)

roleoroleo, 2020-05-13:
What do you mean with off?
Which services do you want to switch off?

fredmassa, 2020-05-13:
The camera itself. My idea is to turn on the camera when the alarm is armed and turn off the camera when the alarm is disarmed

fredmassa, 2020-05-13:
Talking about services... maybe switching on/off these 2 is enough
[image]

roleoroleo, 2020-05-13:
Actually, the "Switch on/off the camera" swithes only Yi services.
If you use RTSP, this switch doesn't change anything.
Check out the wiki about services.

fredmassa, 2020-05-13:
ok so I cannot turn on/off the camera using services like I can do from the Yi app (the camera turn off but remains connected to the wifi so you can turn on it when you need). 

I'll use a smart switch to turn on/off the electricity ;-)

roleoroleo, 2020-05-13:
The switch "Switch on/off the camera" has exactly the same effect as the app switch: it stops the Yi services.
But RTSP is not a Yi service.
You can use camera_settings.sh to switch off the cam and service.sh to start/stop RTSP.

fredmassa, 2020-05-14:
how can i call the camera_settings.sh and the service.sh?
Because i can use ftp to connect to the camera but i can't find the camera_settings.sh or service.sh files

roleoroleo, 2020-05-14:
See the wiki about web services.

fredmassa, 2020-05-15:
Perfect, many thanks!

closed 2020-05-15 by fredmassa