roleoroleo/yi-hack-Allwinner #34: Update to 0.1.6 runs better, but with yi watermark  on rstp

digitalchou, 2020-05-13:
HI, roleoroleo:

After 0.1.6 online update rtsp runs better, but a yi watermark is shown over the video on rtsp.

I have a suggestion too, but i don´t know if its posible:
maybe a switch to disable autoswitch to night mode?
 Many thanks.

roleoroleo, 2020-05-13:
> After 0.1.6 online update rtsp runs better, but a yi watermark is shown over the video on rtsp.

The watermark cannot be removed. You can use your own image if you want but you cannot remove it.
The bmp is 24 bit, the system doesn't support alpha channel and if you try to use a bmp 1x1 pixels the system crashes.

> I have a suggestion too, but i don´t know if its posible:
> maybe a switch to disable autoswitch to night mode?

If I remember correctly the app doesn't have a switch to disable this feature.
I don't know how to discover it, if it exists.

> And a question too: the new option to swap, may benfit rtsp stream?

Maybe, but I have not done extensive tests about.
You could try.

digitalchou, 2020-05-13:
Thanks roleoroleo:
"You can use your own image if you want but you cannot remove it.
The bmp is 24 bit, the system doesn't support alpha channel and if you try to use a bmp 1x1 pixels the system crashes."
ok, I understand I have to search for the watermark BMP and replace with an identical one in size and 24 bit dept, and without alpha channel, it`s correct?

No, the app dont have that option to turn off night mode. 
Reading the post where you explain you cant modifi video encoder options,
I understand that some part of the firmware is "closed" and you can only take
things that it do, or dont take it. 

Well, I will test the swap otion and post the results over rstp.

Great work and many thanks!

pbanj, 2020-05-14:
The file is in /home/app it is named main.bmp there is also sub.bmp

digitalchou, 2020-05-14:
thanks pbanj, i¨ll try

MrPatate87, 2020-05-15:
Can you share your watermark when your done please?

pbanj, 2020-05-17:
This is what I'm using https://cdn.discordapp.com/attachments/399670410565910529/711474097208688640/main.bmp
I just made it white.

closed 2020-07-14 by roleoroleo