roleoroleo/yi-hack-Allwinner #35: Snapshot function delivers the same image multiple times

skdnp, 2020-05-14:
I noticed the behaviour with 0.1.5, but still present in 0.1.6.
Calling the snapshot function via the web interface or also directly via /home/yi-hack/bin/imggrabber multiple times results in the same output. It seems the file /dev/shm/fshare_frame_buf remains static after being once initialized.

I verified with `ssh root@Yi-IP "/home/yi-hack/bin/imggrabber -o stdout" |md5sum` and the hash stays constant over multiple calls.

roleoroleo, 2020-05-14:
It depends on the time between calls.

When the system starts, rmm fills fshare_frame_buf. If rtsp works, the buffer is updated correctly.
Then there is framefinder. It scans the buffer, extracts the addresses of the i-frames and write them in the file /tmp/iframe.idx
Finally, imggrabber reads iframe.idx, extract the frame from the buffer and convert it to jpg.

There is 1 i-frame every 2 seconds, so if you call a snapshot multimple times within 2 seconds, you will get the same picture. But after 2 seconds you should get a new image.
If it doesn't work check if frame.idx changes and check if fshare_frame_buf changes (the first 12 bytes are enough).

skdnp, 2020-05-14:
Very occasionally it does get an updated frame, but it manages to output the same buffer for hours here. All this while rtsp stream is actually streaming.
I just re-checked. (the ~50 extra seconds between the attempts are due to a throttled uplink)

```
while true; do date ;ssh root@Yi-cam 192.168.0.131 "cat /dev/shm/fshare_frame_buf"|md5; sleep 10; done
Do 14 Mai 2020 10:55:23 CEST
20beb791d30be2b91038dce918680aae
Do 14 Mai 2020 10:56:18 CEST
20beb791d30be2b91038dce918680aae
Do 14 Mai 2020 10:57:08 CEST
20beb791d30be2b91038dce918680aae
Do 14 Mai 2020 10:58:04 CEST
20beb791d30be2b91038dce918680aae
Do 14 Mai 2020 10:59:00 CEST
20beb791d30be2b91038dce918680aae
Do 14 Mai 2020 10:59:56 CEST
20beb791d30be2b91038dce918680aae
```

roleoroleo, 2020-05-14:
Probably framefinder is not working as expected.
Check /tmp/iframe.idx

skdnp, 2020-05-14:
Update: I can only replicate the problem on 1 of my 2 rather pretty identical models. Base Version 8.2.0.0A_201912121707 shows the problem, 8.2.0.0A_201912270941 seems to do fine. Yi-cloud offered an update as far as I remember, guess I'll try?

roleoroleo, 2020-05-14:
You could try.

skdnp, 2020-05-28:
Well, update doesn't proceed somehow, so no telling if it would solve this issue.
No errors or warnings from the Yi-app, all the dialogues and the final reboot taking place, yet no version change.

roleoroleo, 2020-05-28:
Try to copy the update file manually.
http://download.eu.xiaoyi.com/yifirmware/smarthomecam/familymonitor-y20ga/8.2.0.0A_201912270941home_y20gam

roleoroleo, 2020-07-14:
Issue without response for 14 days.
Closed.

closed 2020-07-14 by roleoroleo

roleoroleo, 2020-07-16:
I found the problem: when you make a snapshot, the system goes out of run and rmm is killed.
After that, the frame buffer is not updated.

https://github.com/roleoroleo/yi-hack-Allwinner/issues/5
Proably a swap file solve the problem.

aalbul, 2020-07-16:
Hi, sorry, was busy to test the steps you proposed earlier.
I will try the swap file approach though i think i need an SD card for this. Is it possible to completely disable video recording so my SD card will not die too fast?
I didn't find that option. Seems like camera will write them anyway....?

roleoroleo, 2020-07-16:
Are you using cloud or not?

aalbul, 2020-07-16:
No. Cloud is disabled

aalbul, 2020-07-16:
Hm. i think i am a bit confused by "Disable Cloud", "Recording without Cloud" and "Save video when a motion is detected" options.

Can i conclude, that if "Disable Cloud" is turned ON and "Recording without Cloud" is turned OFF, then "Save video when a motion is detected" has no effect?

Ultimately, i would like to disable video recording completely no matter if there was motion or not.

roleoroleo, 2020-07-16:
I know there are confusion with this flag.

The configuration page enable or disable services (linux programs).
When "Disable cloud" is on, during the boot process, the cam doesn't load all cloud related programs (p2p_tnp, cloud, etc...), so it doesn't connect to the cloud.
When "Recording without cloud" is on, the cam load mp4record, the program that save mp4 video files.

The "Camera Settings" page simply copies some checks from the Yi app to the web interface, to enable/disable original features.

> Can i conclude, that if "Disable Cloud" is turned ON and "Recording without Cloud" is turned OFF, then "Save video when a motion is detected" has no effect?

True.

> Ultimately, i would like to disable video recording completely no matter if there was motion or not.

- "Disable cloud" ON
- "Recording without Cloud" OFF

should be enough.

daton89, 2020-07-25:
I have a camera with firmware 8.2.0.0A_201912270941 and unless I do several snapshots in high res consecutively the camera  works fine.

But the one with firmware 8.2.0.0A_201912121707 is crashing when I do a high res snapshot. 

May this be related this or other bugs I saw? Like #13 or #5 #35

Is there something we can do to fix that problem?

roleoroleo, 2020-07-25:
> I tried to disable all the services and enabled the swap and it apparently works much better, but do you suggest to upgrade the firmware?

I don't know the changelog of the new firmware.
And I didn't try the old version.

> Is there something else we can do to make sure the camera doesn't run out of memory or crashes?

Only switch off all unnecessary services.