roleoroleo/yi-hack-Allwinner #36: Can't use last hack and yi home app

MrPatate87, 2020-05-15:
Hello and thanks for the great hack and all this nex improvements.
I have a problem to use my camera with yi home app when i apply your hack.
Innfact for the app my camera if out of line  so I can't use the app feature and can't stream the video to a google home for exemple.What i'm doing wrong ?
In first time I think it the password authentification but when I restore to default (none) I have no more app connexion.

roleoroleo, 2020-05-15:
Don't check the button "Disable cloud" and you will be able to use Yi app.

MrPatate87, 2020-05-15:
Work great on one of my camera (actually on 1.4 version, but can't connect on the camera with the 1.6 hack version.

scruff-01, 2020-05-15:
Did you delete the camera from the app and add it back? Mine has been finicky with wifi as well and deleting re adding it works. Then apply the .16 again. Thats as far as I've got where it will still work

MrPatate87, 2020-05-15:
Yes i already try this.
I delete the camera from the app, reboot it without micro sd and trying to connect like a new one. When I made the wifi connexion process it can connect to wifi and the hack is still there (can connect to camera configuration via Web interface and have rtsp video-but no audio with 1.6...) but the app can't connect to the camera (Same result if I do this with a micro sd). 
If I power off the camera, this one can't reconnect to wifi and I must re do the connexion process (saying waiting go connect).

pbanj, 2020-06-04:
One of my newer ones was doing that but it stopped after I enabled cloud again, then ran through the setup in the app. Then I was able to disable the cloud app after it told me setup was complete

spamsool, 2020-06-22:
I have just setup the Y31 version of the camera and noticed immediately that the picture quality of y20gam is much worse then the Mstar version y203c. In fact the bitrate is almost 3x higher in the y203 version. Is there anything that could be done about it? Allwinner model has the picture which is just too much compressed and looks horrible.

closed 2020-07-14 by roleoroleo