roleoroleo/yi-hack-Allwinner #37: v0.1.6 - MQTT motion NOT send ON (start)

wadalino, 2020-05-16:
i've configured **MQTT Motion Topic** as 
Topic Suffix : **motion**
Motion Start Message : **on**
Motion Stop Message : **off**
Topic Suffix for video list message : **files**
```
------------------ log # python3 mqtt-data-logger.py -b ?.?.?.? -u ? -P ? -t "yicamdos/#" -t "yicamuno/#" 
Python version is sys.version_info(major=3, minor=5, micro=3, releaselevel='final', serial=0)
Log Directory = mlogs
Log records per log = 5000
Max logs = Unlimited
initialising clients
Logging JSON format
starting storing only changed data
{"topic": "yicamdos/motion_sensor", "time": [2020, 5, 16, 11, 34, 0, 5, 137, 1], "message": "motion_stop"}
{"topic": "yicamdos/motion_files", "time": [2020, 5, 16, 11, 34, 0, 5, 137, 1], "message": {"start": "1970-01-01T01:00:00+0100", "end": "2020-05-16T09:44:01+0200", "files": ["2020Y05M16D09H/44M00S60.mp4", "2020Y05M16D09H/43M00S60.mp4"]}}
{"topic": "yicamdos/motion", "time": [2020, 5, 16, 11, 34, 0, 5, 137, 1], "message": "off"}
{"topic": "yicamdos/files", "time": [2020, 5, 16, 11, 34, 0, 5, 137, 1], "message": {"start": "1970-01-01T01:00:00+0100", "end": "2020-05-16T10:45:30+0200", "files": ["2020Y05M16D10H/45M00S60.mp4", "2020Y05M16D10H/44M27S32.mp4"]}}
{"topic": "yicamuno/motion_sensor", "time": [2020, 5, 16, 11, 34, 0, 5, 137, 1], "message": "motion_stop"}
{"topic": "yicamuno/motion_files", "time": [2020, 5, 16, 11, 34, 0, 5, 137, 1], "message": "{\n\"start\":\"1970-01-01T01:00:00+0100\",\n\"end\":\"2020-05-15T22:16:26+0200\",\n\"files\": ]\n}\n"}
{"topic": "yicamuno/motion", "time": [2020, 5, 16, 11, 34, 0, 5, 137, 1], "message": "off"}
{"topic": "yicamuno/files", "time": [2020, 5, 16, 11, 34, 0, 5, 137, 1], "message": "{\n\"start\":\"1970-01-01T01:00:00+0100\",\n\"end\":\"2020-05-16T11:32:32+0200\",\n\"files\": ]\n}\n"}
{"topic": "yicamuno/files", "time": [2020, 5, 16, 11, 34, 43, 5, 137, 1], "message": "{\n\"start\":\"1970-01-01T01:00:00+0100\",\n\"end\":\"2020-05-16T11:33:33+0200\",\n\"files\": ]\n}\n"}
{"topic": "yicamuno/files", "time": [2020, 5, 16, 11, 35, 45, 5, 137, 1], "message": "{\n\"start\":\"1970-01-01T01:00:00+0100\",\n\"end\":\"2020-05-16T11:34:35+0200\",\n\"files\": ]\n}\n"}
{"topic": "yicamuno/files", "time": [2020, 5, 16, 11, 36, 48, 5, 137, 1], "message": "{\n\"start\":\"1970-01-01T01:00:00+0100\",\n\"end\":\"2020-05-16T11:35:38+0200\",\n\"files\": ]\n}\n"}
```
```
------------------ log MQTT Home Assistant
Mensaje 35 recibido en yicamuno/motion a las 11:23: off QoS: 0 - Retain: false
Mensaje 34 recibido en yicamuno/files a las 11:22: {"start":"1970-01-01T01:00:00+0100", "end":"2020-05-16T11:21:32+0200", "files": ]} QoS: 0 - Retain: false
Mensaje 33 recibido en yicamuno/motion a las 11:22: off QoS: 0 - Retain: false
Mensaje 32 recibido en yicamuno/files a las 11:21: {"start":"1970-01-01T01:00:00+0100", "end":"2020-05-16T11:20:30+0200", "files": ]} QoS: 0 - Retain: false
Mensaje 31 recibido en yicamuno/motion a las 11:21: off QoS: 0 - Retain: false
```

roleoroleo, 2020-05-16:
Start mqttv4 manually and check the log in stdout.

wadalino, 2020-05-16:
how can run manually? :(

roleoroleo, 2020-05-16:
Login with ssh.
```
killall mqttv4
/home/yi-hack/bin/mqttv4
```

wadalino, 2020-05-16:
```
root@yicamdos:~# killall -9 mqttv4
root@yicamdos:~# /home/yi-hack/bin/mqttv4 
Error loading shared library libmosquitto.so.1: No such file or directory (needed by /home/yi-hack/bin/mqttv4)
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_connack_string: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_publish: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_publish_callback_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_connect_callback_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_log_callback_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_username_pw_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_lib_cleanup: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_new: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_connect: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_strerror: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_disconnect_callback_set: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_loop: symbol not found
Error relocating /home/yi-hack/bin/mqttv4: mosquitto_lib_init: symbol not found
```

roleoroleo, 2020-05-16:
Login with ssh.
```
killall mqttv4
. /home/yi-hack/script/env.sh
/home/yi-hack/bin/mqttv4
```

wadalino, 2020-05-16:
```
root@yicamuno:/home/yi-hack# /home/yi-hack/bin/mqttv4
Starting mqttv4 v0.0.3
Trying to connect... Client yicamuno sending CONNECT
Client yicamuno received CONNACK (0)
. connected!
CALLBACK MOTION STOP
Error: Invalid input. Does your topic contain '+' or '#'?
CALLBACK MOTION STOP
Error: Invalid input. Does your topic contain '+' or '#'?
SENDING FILES LIST
Cannot open directory: 2020Y05M16D11H
Error: Invalid input. Does your topic contain '+' or '#'?
Client yicamuno sending PINGREQ
Client yicamuno received PINGRESP
SENDING FILES LIST
Cannot open directory: 2020Y05M16D11H
Error: Invalid input. Does your topic contain '+' or '#'?
CALLBACK MOTION STOP
Error: Invalid input. Does your topic contain '+' or '#'?
SENDING FILES LIST
Cannot open directory: 2020Y05M16D11H
Error: Invalid input. Does your topic contain '+' or '#'?
Client yicamuno sending PINGREQ
Client yicamuno received PINGRESP
CALLBACK MOTION STOP
Error: Invalid input. Does your topic contain '+' or '#'?
CALLBACK MOTION STOP
Error: Invalid input. Does your topic contain '+' or '#'?
SENDING FILES LIST
Cannot open directory: 2020Y05M16D11H
Error: Invalid input. Does your topic contain '+' or '#'?
```

wadalino, 2020-05-18:
After lot probes it resolvs.
I was update latest firmware yi-cam manualy
Then apply hack
And just

closed 2020-05-18 by wadalino

matteopiccioni, 2020-06-23:
Hello, I have the same problem
I receive only stop messages from mqtt

My config
```
MQTT_IP=192.x.x.x
MQTT_PORT=1883

MQTT_CLIENT_ID=yi-cam

MQTT_USER=myuser
MQTT_PASSWORD=mypwd

MQTT_PREFIX=home-assistant/ingresso

# -----------------------------------------------------------------------------
# Set the topics where the messages will be published
# The final topic will be MQTT_PREFIX/TOPIC_MOTION
# -----------------------------------------------------------------------------

TOPIC_MOTION=motion_detection
TOPIC_MOTION_FILES=motion_files
TOPIC_BABY_CRYING=baby_crying

# -----------------------------------------------------------------------------
# Set the topics messages
# -----------------------------------------------------------------------------

MOTION_START_MSG=ON
MOTION_STOP_MSG=OFF
BABY_CRYING_MSG=crying

# -----------------------------------------------------------------------------
# Other settings
# -----------------------------------------------------------------------------

MQTT_KEEPALIVE=120
MQTT_QOS=1
MQTT_RETAIN=1
```

In Home Assistant I enable listening for 
   home-assistant/ingresso/motion_detection 
and only OFF messages comes:
```
Message 5 received on home-assistant/ingresso/motion_detection at 9:56 AM:
OFF
QoS: 0 - Retain: false
Message 4 received on home-assistant/ingresso/motion_detection at 9:54 AM:
OFF
QoS: 0 - Retain: false
Message 3 received on home-assistant/ingresso/motion_detection at 9:51 AM:
OFF
QoS: 0 - Retain: false
```

Using ssh I send this commands:
```
killall mqttv4
. /home/yi-hack/script/env.sh
/home/yi-hack/bin/mqttv4
```
The console shows:
```
root@xiaoyi:~# killall mqttv4
root@xiaoyi:~# . /home/yi-hack/script/env.sh
root@xiaoyi:~# /home/yi-hack/bin/mqttv4
Starting mqttv4 v0.0.3
Trying to connect... Client yi-cam sending CONNECT
Client yi-cam received CONNACK (0)
. connected!
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r1, m1, 'home-assistant/ingresso/motion_detection', ... (3 bytes))
Client yi-cam received PUBACK (Mid: 1)
SENDING FILES LIST
Client yi-cam sending PUBLISH (d0, q1, r1, m2, 'home-assistant/ingresso/motion_files', ... (1635 bytes))
Client yi-cam received PUBACK (Mid: 2)
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r1, m3, 'home-assistant/ingresso/motion_detection', ... (3 bytes))
Client yi-cam received PUBACK (Mid: 3)
SENDING FILES LIST
Client yi-cam sending PUBLISH (d0, q1, r1, m4, 'home-assistant/ingresso/motion_files', ... (1635 bytes))
Client yi-cam received PUBACK (Mid: 4)
CALLBACK MOTION STOP
Client yi-cam sending PUBLISH (d0, q1, r1, m5, 'home-assistant/ingresso/motion_detection', ... (3 bytes))
Client yi-cam received PUBACK (Mid: 5)
```

Should ON messages comes too?

roleoroleo, 2020-06-23:
Try the solution above.

- Reset the cam with the official procedure. - Upgrade the yi fw if available.
- Reconfigure the cam with the official app.
- Apply again the hack.

matteopiccioni, 2020-06-23:
Thanks, the solution was collapsed :)
It was enough to reset and reconfigure the cam