roleoroleo/yi-hack-Allwinner #40: Adjust bitrate

zZjessahZz, 2020-05-16:
Hi roleoroleo,
Thank you for bringing yi-hack to the allwinner chipset!

Do you think there is a possibility to adjust the bitrate of video stored in the sdcard?
I wasn't able to find a config file on the camera related to that.

Thank you for your insight!

roleoroleo, 2020-05-16:
No, sorry. It's not possible.
We don't have control about the encoding phase.

zZjessahZz, 2020-05-16:
Oh that's too bad. Is this because this part is hidden in a code blob? (I mean like the little part of the SOC firmware on the raspberry pi thats not open source.)

roleoroleo, 2020-05-17:
Yes, it's inside the rmm application, that is closed source.

zZjessahZz, 2020-05-17:
Okay I understand, thank you for your explaination roleoroleo.
Have a nice day !

closed 2020-05-17 by zZjessahZz