roleoroleo/yi-hack-Allwinner #42: Camera continuously rebooting after hack when replugged on power outlet

jfallot, 2020-05-18:
Hello,

After having configured the camera with the Yi Home App, I installed the version 0.1.6 of the Allwinner hack (the camera was connected to the USB plug of the computer):
everything was working fine: Http access to the configuration page on port 8880, RTSP stream through port 554 and the cam was still visible from the Yi Home App.

1. I unplugged the camera from the PC to plug it directly to the power outlet using the stock power unit.
2. Then the camera was endlessly rebooting (Steady Orange light -> Blinking Blue light -> Steady Orange -> Blinking blue, etc...). The microSD containing the AllWinner firmware was still in the device.
3. When reconnecting the camera to the PC, it was booting normally and working again (Yi Home App, RTSP access, SSH, SCP) but the Http page on port 8080 is no longer accessible
I repeated steps 2. and 3. several times and got the same results.
I tried to reset the device or removing the microSD from the device, without any improvement.

Can you provide guidance to troubleshoot and fix that issue?

Thanks

roleoroleo, 2020-05-18:
Probably the upgrade process doesn't finish correctly.
Try to repeat again from the beginning reformatting sd card.

jfallot, 2020-05-18:
Thanks for your prompt reply, this helped to get back the hacked camera working when connected to the computer.
I also identified the source of this endless rebooting issue when plugged to the power outlet: it was caused by a faulty USB extension cable.
Issue is then closed, no problem with the hack, sorry for the inconvenience!

closed 2020-05-18 by jfallot