roleoroleo/yi-hack-Allwinner #44: High Quality RTSP does not work in the daytime

knguyen298, 2020-05-27:
I have two Yi Home 1080P cameras, both of which are the same "9FUS" version. 

The one that is having issues is on base version 8.2.0.0A_201912270941. During the night, the high quality feed works fine. However, during the day, the high quality feed doesn't work, and when I play it though VLC, I can hear the audio, but I don't get video. The low quality feed works fine; both audio and video work. 

I have tried to turn off the IR light setting, but it doesn't seem to fix anything. This camera was recently converted, so 0.1.6 was the first version installed on it.

My other camera is on base version 8.2.0.0A_201912121707. I've been using the hack on this camera for quite a while, so it was updated to 0.1.6 within the settings. The camera has no issues with the high quality RTSP feed working in the day or night.

knguyen298, 2020-05-28:
Update: I just verified that the high quality feed is only functional when the IR light is on. The feed shut off this morning once it got light enough to switch off IR (confirmed with the other camera I have in a similar location). Then, the high quality feed turned on again earlier this evening, when IR is needed (again, confirmed with the other camera). Only the audio is working on the high quality feed.

I also replicated the upgrade path that the fully functional camera followed: Installing 0.1.4, then updating to 0.1.6. I wiped the install clean prior to starting this process; I followed the procedure as listed in issue #12, also wiping the 'yi-hack-v4' in the same directory that 'yi-hack' was in. 

If anyone else is on the same base version (8.2.0.0A_201912270941), is your high quality stream working? Also, is there any logs I can check to see if any errors are being thrown?

roleoroleo, 2020-05-30:
Please check if the stream is available with yi app.

knguyen298, 2020-05-30:
Yes, the "HD" stream works on the Yi app. I have had the HD stream pulled up on the app, and tried to view the RTSP stream using VLC, and I will only get audio from the stream.

roleoroleo, 2020-06-16:
If you disable audio, does the stream work?

luminouw, 2020-06-18:
Hello,

just got my hands on a 9FUS version as well (Base Version 8.2.0.0A_201912270941), and decided to give the YiHack treatment (flashed 0.1.6). As @roleoroleo mentionned I am facing the same issue except it affects LOW quality on my side and I'm unable to view anything when trying to access the High Quality feed (regardless if IR led is on or off).

I'll perform some more tests (with audio on/off, set on high or low video feed) tonight and post the results here.

Viewing the video through the app works flawlessly and the snapshot function works as well.

knguyen298, 2020-06-22:
Turning off the audio does nothing. I just tried it again today after leaving it alone for ~2 weeks, and same thing. When the audio is on however, I can hear the audio play through relatively clearly.

roleoroleo, 2020-06-22:
I checked all the scripts and the rtsp daemon.
I don't understand why the system doesn't work during day.

Could be a memory problem?. Did you try to enable swap file on sd?

pniewiadowski, 2020-06-27:
I have sam problem.
SD swap enabled.

vmendivil, 2020-06-27:
I face the same issue, in my case, in the YI app, when trying to see the recordings on the SD card, it shows at what time there were some activity but when trying to see the recordings it just skips right away up to live stream. I pulled out the SD card, plug it in on the computer and the recordings are there, I can see the on PC but the YI app, even though it recognized them, won't play them.

roleoroleo, 2020-06-27:
I need more details to understand yhe problem.
Try to start the grabber with debug options.

vmendivil, 2020-06-27:
How do I start the grabber with debug options?

roleoroleo, 2020-06-27:
Kill the running processes, kill wd_rtsp also.
Then run in shell:
h264grabber -d | rRTSPServer

vmendivil, 2020-06-27:
Will do once I'm back. 

Additional update. The fact Yi app wasn't able to reproduce recordings was because timezone was configure in the yi-hack. As soon as I removed the timezone (that matched what I have configure on the yi app), I was able to start reproducing videos again, fun part, if I take a snapthot in the yi-hack web interfacem, timestamp is UTC since I removed the timezone.

The only issue remaining, as the title states, is that HD stream doesn't work all the time. SD stream works.

pniewiadowski, 2020-06-27:
PS. Just removed the timezone to see if that solve my problem with yi app and the recordings (same issue as @vmendivil has) and after restart, my stream is back online. I'll keep an eye on this and if the problem occur I will pass you the grabber data.

benooye, 2020-06-29:
First really thanks for this work.
I have the same issue here : Firmware Version | 0.1.6 / Base Version | 8.2.0.0A_201912270941 / Model Suffix | y20gam
Everything is working fine except the high quality RTSP stream is only functional when the IR light is on.
I have tried to remove timezone info and NTP, but unfortunately I have still the same issue...
I will try to get back to stock firmware and then update again and stick to default configuration.

pniewiadowski, 2020-06-29:
> Kill the running processes, kill wd_rtsp also.
> Then run in shell:
> h264grabber -d | rRTSPServer

Seems it doesn't work for me:
```
root@outdoor:~# h264grabber -d | rRTSPServer
-ash: rRTSPServer: not found
-ash: h264grabber: not found

```

roleoroleo, 2020-06-29:
`. /home/yi-hack/script/env.sh`

pniewiadowski, 2020-06-29:
This is what I got:

```root@outdoor:~# h264grabber -d | rRTSPServer
-ash: h264grabber: not found
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size

"ch0_0.h264" stream, from memory
Play this stream using the URL "rtsp://192.168.0.122/ch0_0.h264"
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
```

roleoroleo, 2020-06-29:
?!?
h264grabber not found?
Try this:
`ls -la /home/yi-hack/bin`
and this
`df`

pniewiadowski, 2020-06-30:
```
root@outdoor:~# ls -la /home/yi-hack/bin/
drwxr-xr-x    2 root     root             0 Jun 22 16:52 .
drwxr-xr-x    9 root     root             0 Jun 22 16:52 ..
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 arch -> busybox
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 base64 -> busybox
-rwxr-xr-x    1 root     root        290152 Jun 22 16:52 busybox
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 cat -> busybox
lrwxrwxrwx    1 root     root            31 Jun 22 16:52 dbclient -> /home/yi-hack/bin/dropbearmulti
lrwxrwxrwx    1 root     root             7 Jun 22 16:51 dmesg -> busybox
lrwxrwxrwx    1 root     root            31 Jun 22 16:51 dropbearconvert -> /home/yi-hack/bin/dropbearmulti
lrwxrwxrwx    1 root     root            31 Jun 22 16:52 dropbearkey -> /home/yi-hack/bin/dropbearmulti
-rwxr-xr-x    1 root     root        219980 Jun 22 16:52 dropbearmulti
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 echo -> busybox
-rwxr-xr-x    1 root     root          5988 Jun 22 16:52 framefinder
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 fsync -> busybox
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 hostname -> busybox
-rwxr-xr-x    1 root     root       1439948 Jun 22 16:52 imggrabber
lrwxrwxrwx    1 root     root             7 Jun 22 16:51 iostat -> busybox
-rwxr-xr-x    1 root     root          9904 Jun 22 16:52 ipc_cmd
lrwxrwxrwx    1 root     root             7 Jun 22 16:51 more -> busybox
-rwxr-xr-x    1 root     root         35824 Jun 22 16:52 mosquitto_pub
-rwxr-xr-x    1 root     root         38672 Jun 22 16:52 mosquitto_sub
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 mpstat -> busybox
-rwxr-xr-x    1 root     root         12364 Jun 22 16:52 mqttv4
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 netstat -> busybox
-rwxr-xr-x    1 root     root       6238224 Jun 22 16:51 onvif_srvd
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 ping -> busybox
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 printenv -> busybox
-rwxr-xr-x    1 root     root          4836 Jun 22 16:51 proccgi
-rwxr-xr-x    1 root     root          6820 Jun 22 16:52 proxychains4
-rwxr-xr-x    1 root     root        345080 Jun 22 16:51 rRTSPServer
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 resume -> busybox
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 rmdir -> busybox
lrwxrwxrwx    1 root     root            31 Jun 22 16:51 scp -> /home/yi-hack/bin/dropbearmulti
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 setpriv -> busybox
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 sleep -> busybox
lrwxrwxrwx    1 root     root            31 Jun 22 16:52 ssh -> /home/yi-hack/bin/dropbearmulti
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 su -> busybox
lrwxrwxrwx    1 root     root             7 Jun 22 16:52 sync -> busybox
root@outdoor:~# df
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776     10284      1492  87% /home
tmpfs                    32768        96     32672   0% /tmp
/dev/mmcblk0p1        31158432  11567488  19590944  37% /tmp/sd
root@outdoor:~#
```

roleoroleo, 2020-06-30:
Ok, sorry.
I made a mistake on the version.
The right command is only `rRTSPServer`

pniewiadowski, 2020-06-30:
This gave me similar output:

```
root@outdoor:~# rRTSPServer
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size

"ch0_0.h264" stream, from memory
Play this stream using the URL "rtsp://192.168.0.122/ch0_0.h264"
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
frame size exceeds buffer size
```

roleoroleo, 2020-06-30:
Try this alpha version:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/55

closed 2020-08-29 by roleoroleo