roleoroleo/yi-hack-Allwinner #45: Wifi often lost; SD card full

Xandrios, 2020-05-27:
I've noticed two weird things.

Over the last few days I've left my play-around-with camera unplugged. When I plugged it back in yesterday it had lost the Wifi settings and required to be re-paired with the app. Considered it a fluke until the exact same thing happened again today. However in both cases after re-pairing I was able to disconnect it from power without issues...without it losing the settings.

Do we know what causes the Wifi settings to be lost? When not connected I cannot check the state of the /tmp/wpa_supplicant.conf file. Who is responsible for putting that file there? (assuming /tmp/ is empty at boot and the file needs to be put there at startup). Is there some master version of that file somewhere that can also be reset using the pinhole button?

Secondly I noticed the SD card to be almost full, while I had set it to always keep 10% free. Looking at the system.sh script this does not work for me...at all. Does it work for others?

For me cron does not work: 
`/usr/sbin/crond -c /var/spool/cron/crontabs/
-ash: /usr/sbin/crond: not found
`

And also the cleanup script itself does not seem to work, as it relies on 'df -h'...while the df command that is installed does not seem to know that -h option?

`/home/yi-hack/script/clean_records.sh 10
90
df: unrecognized option: h
`

roleoroleo, 2020-05-28:
> 
> 
> Who is responsible for putting that file there? (assuming /tmp/ is empty at boot and the file needs to be put there at startup).

It's not clear but probably it's a binary from the original yi firmware.

> Is there some master version of that file somewhere that can also be reset using the pinhole button?

No.

> Secondly I noticed the SD card to be almost full, while I had set it to always keep 10% free. Looking at the system.sh script this does not work for me...at all. Does it work for others?

There is a problem. Probably I accidentally overwrote this file with the MStar versione.
I will correct it in the next release.
If you want to correct it now, change the crond path with "/home/yi-hack/usr/bin" and remove -h from df command

Xandrios, 2020-06-20:
Thanks. No problems, I have created a cleanup script specifically for my environment since I have additional directories to cleanup (record, record_motion, record_snapshots, etc).

closed 2020-07-14 by roleoroleo