roleoroleo/yi-hack-Allwinner #46: Snapshot with high resolution starts showing the same image after a while

aalbul, 2020-05-28:
Hello,
Thanks for a great work on hacking this camera!
It works well for me most of the time, though i'm experiencing some weird issue with image snapshot in high resolution when camera runs for a day or so.

I use it with Home Assistant by defining it as generic camera in such a way:

```yaml
- platform: generic
  name: Living Room Panoramic Camera
  stream_source: rtsp://192.168.2.54:554/ch0_0.h264
  still_image_url: http://192.168.2.54:8080/cgi-bin/snapshot.sh?res=high&watermark=yes
  verify_ssl: false
```

As you can see, i set res to "high".

In lovelace i define it like so:

```yaml
- aspect_ratio: 50%
  camera_view: live
  entity: camera.living_room_panoramic_camera
  type: picture-entity
```

With such a definition Home Assistant query camera's snapshot endpoint now and then when you are looking at it via UI.

After ~day camera starts showing outdated image that never change (watermark changes though and rtsp works as well). The only way to fix it is to reboot the camera.
If i switch back to "low" res, it works without any issues.

Is it a bug or just some misconfiguration on my side?

P.S.: Is there any chance to implement "mid" res suppose? Low res resolution is too small, high res is fine, but quite slow.

Regards, Alex

aalbul, 2020-05-28:
Forgot to mention: i tried enabling / disabling SWAP - the same result

roleoroleo, 2020-05-28:
I think it could be a bug.
I will check it.

About mid res, unfortunately it's not possible.

PS Did you try to define the camera as onvif in ha? With the release 0.110 it works very well.

aalbul, 2020-05-28:
Thanks for your reply. I did try onvif integration, but to be fair, it is quite slow for me.
Possible, it's because of my Raspberry Pi 3B+ is not powerful enough for decoding / encoding streams.
It's one of the reasons why i use static images most of the time.

roleoroleo, 2020-05-30:
I checked it on my cam after 2 days and the snapshot is ok.
When the problem happens again, please check these things:

1. Check if framefinder process is running (with ps command).
2. Check if the content of the file /tmp/iframe.idx changes (with hexdump command).

roleoroleo, 2020-07-14:
Issue without response for 14 days.
Closed.

closed 2020-07-14 by roleoroleo

roleoroleo, 2020-07-16:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/35#issuecomment-659207376