roleoroleo/yi-hack-Allwinner #47: add way to change starting port

pbanj, 2020-05-29:
can you add a way for us to set the port before installing the firmware? i have stuff that uses port 8080 so i have to disable it every time i want to set up a camera.

roleoroleo, 2020-05-30:
If you update with ota feature, the cam saves and restores configuration. When the system reboots you don't have to reconfigure it.

If you use update files on sd, you can follow this procedure:
1. Create /tmp/sd/fw_upgrade directory
2. Copy /home/yi-hack/etc/*.conf in /tmp/sd/fw_upgrade/
3. Copy /home/yi-hack/etc/hostname in /tmp/sd/fw_upgrade/
4. Copy the update files
5. Reboot the cam

The new firmware will find backup files and will restore it.

pbanj, 2020-05-30:
Oh I'm not talking about updating the cameras. I'm talking about if you get more and are setting them up

roleoroleo, 2020-05-30:
You could still use the method above.
You could create directory and files using a PC, maybe using a backup from another CAM as a template.

What service are you using on port 8080 of the cam?

pbanj, 2020-05-30:
That would work, I'll give it a try next time. My music server uses port 8080.

closed 2020-05-30 by pbanj