roleoroleo/yi-hack-Allwinner #48: Firmware 0.1.6 only telnet

hocker96, 2020-05-30:
I had update from 0.1.4 to 0.1.6,

after this, only telnet work.

Then I install 0.1.6 after reset, only telnet is open.

Install 0.1.4, then it work's

roleoroleo, 2020-05-30:
Is the web interface running?

hocker96, 2020-05-31:
0.1.4 yes
> root@yi-b82c:/home/app/script# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
  179 root         0 SW   [oom_reaper]
  180 root         0 SW<  [writeback]
  182 root         0 SW   [kcompactd0]
  183 root         0 SW<  [crypto]
  184 root         0 SW<  [bioset]
  186 root         0 SW<  [kblockd]
  224 root         0 SW   [sys_user]
  231 root         0 SW   [kworker/0:1]
  232 root         0 SW<  [cfg80211]
  239 root         0 SW<  [watchdogd]
  253 root         0 SW<  [spi0]
  263 root         0 SW   [kswapd0]
  348 root         0 SW<  [bioset]
  353 root         0 SW<  [bioset]
  358 root         0 SW<  [bioset]
  363 root         0 SW<  [bioset]
  368 root         0 SW<  [bioset]
  373 root         0 SW<  [bioset]
  378 root         0 SW<  [bioset]
  383 root         0 SW<  [bioset]
  402 root         0 SW   [irq/302-sunxi-m]
  404 root         0 SW   [irq/166-sdc0 cd]
  405 root         0 SW   [irq/303-sunxi-m]
  446 root         0 SW   [kworker/u2:2]
  463 root         0 SW<  [kworker/0:1H]
  466 root         0 SW<  [bioset]
  467 root         0 SW   [mmcqd/0]
  482 root         0 SWN  [jffs2_gcd_mtd3]
  487 root       980 S    -/bin/sh
  528 root         0 SW   [ksdioirqd/mmc1]
  531 root         0 SW   [RTW_XMIT_THREAD]
  532 root         0 SW   [RTW_CMD_THREAD]
  533 root         0 SW   [RTWHALXT]
  534 root       680 S    ./log_server
  535 root       984 S    ./dispatch
  541 root     47448 S    ./rmm
  575 root       988 S    /usr/sbin/telnetd
  826 root       700 S    ./mp4record
  837 root       844 S    httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
  848 root       844 S    pure-ftpd (SERVER)
  862 root       876 S    dropbear -R -B
  885 root       844 S    ntpd -p pool.ntp.org
  902 root       984 S    {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
  936 root      8036 S    onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Yi Hack --manufacturer Yi --ifs wlan0 --port 80 --scope onvif://www.onvif.org/Profile/S --name Profile_0 --width 1920 --height 1080 --url rtsp://%s/
  937 root      2420 S    framefinder
  975 root      1400 S    /home/base/tools/wpa_supplicant -c/tmp/sd/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B
 1019 root       980 S    udhcpc -i wlan0 -b -s /home/app/script/default.script
 1613 root      4048 S    rRTSPServer
 1765 root       980 S    -ash
 8328 root       980 S    sleep 10
 8357 root       980 R    ps
>

0.1.6 was no

but after the 10th test it working upgrade from 0.1.4 to 0.1.6

hocker96, 2020-05-31:
Em no... 0.1.6 upgrade over webinterface was not succesfull I show them later

roleoroleo, 2020-05-31:
The log of your ps shows that dropbear is running.
You should be able to login with ssh.
Event rtsp server is running.
Check your network connection, firewall, etc...

hocker96, 2020-05-31:
I can't Upgrade from 0.1.4 to 0.1.6. There ist an issus. 

By 0.1.6 was only telnet start. No dropbear, no httpd, no ftp. Nothing

roleoroleo, 2020-06-01:
If the cam is the same described here https://github.com/roleoroleo/yi-hack-Allwinner/issues/49 the problem could be "no space left on home partition".
Check if there are core files on /home/app.

roleoroleo, 2020-07-14:
Issue without response for 14 days.
Closed.

closed 2020-07-14 by roleoroleo