roleoroleo/yi-hack-Allwinner #49: Camera do not save Wifi config, wpa_suplicant.conf from SD?

hocker96, 2020-05-30:
My camera do not connect to my wifi after reboot, they want to scan the QR code. Tested with 0.1.4 and 0.1.6.

Is it posible to make as option to read and copy the wpa_suplicant.conf from SD card?

roleoroleo, 2020-05-30:
You can read wpa_supplicant but you can't write it.
Wifi configuration is saved on flash (a raw partition), wpa_supplicant is "only" a copy.

Xandrios, 2020-05-30:
Having the same issue with WiFi in #45. I did not disable the Yi cloud as I'd like to have the ability to use the Yi app to Live stream. Every now and then it does lose the Wifi config though...which makes me hesitant to actually install the camera at a hard-to-reach-place for security purposes.

I was wondering though...If we, at boot using startup.sh, forcefully copy/set a Wifi config? Possibly even kill any running instances of wpa_suplicant and start a new one with our custom wpa_suplicant.conf file as argument?

Only thing that I am not sure about is the camera state when not connected to the WiFi. Do all the services come up like normal? Or does it go into some other state and first needs to be fed before working normally?

roleoroleo, 2020-05-30:
> I was wondering though...If we, at boot using startup.sh, forcefully copy/set a Wifi config? Possibly even kill any running instances of wpa_suplicant and start a new one with our custom wpa_suplicant.conf file as argument?

It could be possible but we need ESSID and password.
It depends on whether the cam lost them or not.

> Only thing that I am not sure about is the camera state when not connected to the WiFi. Do all the services come up like normal? Or does it go into some other state and first needs to be fed before working normally?

I don't know.
Nevertheless all the tests I have done, it has never happened to me.
The same problem happens in the Mstar model.

hocker96, 2020-05-31:
I had buy the camera to use without cloud.

I have copy the wpa_suplicant.conf from /tmp to /tmp/sd

and change the /home/app/script# vi wificonnect.sh

to

`/home/base/tools/wpa_supplicant -c/tmp/sd/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B;
`

it works, but it's dangoures, this file are be no reset ore overwrite with stage. An the camera want to scan a QRCode and start then wificonnect.sh, with my file.

the wpa_supplicant.conf is 777, can you write a script to copy the file from sd and stop QRCode scanning if exsist and restart wlan, so you do not have a QRCode

`drwxrwxrwt    6 root     root           460 May 31 01:19 .
drwxrwxr-x   14 root     root           195 Oct 30  2019 ..
-rw-r--r--    1 root     root             0 Apr 17 20:16 .count
lrwxrwxrwx    1 root     root            23 Jan  1  1970 audio -> /home/app/audio_file/us
-rw-r--r--    1 root     root          1587 May 31 01:18 debug_alarm.txt
-rw-r--r--    1 root     root             0 Jan  1  1970 debug_oss.txt
-rw-r--r--    1 root     root             0 Jan  1  1970 debug_p2p.txt
-rw-r--r--    1 root     root            43 May 31 01:18 dns1
-rw-r--r--    1 root     root             2 May 31 01:18 done
-rw-r--r--    1 root     root            13 May 31 01:18 gw1
-rw-r--r--    1 root     root             0 Jan 19  2016 httpd.conf
-rw-r--r--    1 root     root            48 May 31 01:19 iframe.idx
drwxr-xr-x    2 root     root            40 Jan  1  1970 log
-rw-r--r--    1 root     root          9876 May 31 01:18 log.txt
-rw-r--r--    1 root     root             0 Jan  1  1970 log_oss.txt
srwxr-xr-x    1 root     root             0 Jan  1  1970 logsock
-rw-r--r--    1 root     root            14 May 31 01:18 mask
-rwxr-xr-x    1 root     root          2508 Jan  1  1970 mmap.info
-rw-r--r--    1 root     root           139 May 31 01:18 resolv.conf
drwxr-xr-x    2 root     root            40 Jan  1  1970 run
drwxr-xr-x    6 root     root         16384 Apr 17 20:16 sd
drwxr-xr-x    3 root     root            60 Jan  1  1970 var
-rwxrwxrwx    1 root     root           166 Apr 17 20:16 wpa_supplicant.conf
`

roleoroleo, 2020-05-31:
Sorry, but I did not understand.
Could you explain me better?

Xandrios, 2020-05-31:
Question is when the "not paired" voice message is played. Is that while the camera has fully start up but somehow does not manage to (re) connect to the Wifi network that has previously been configured? Or does that happen when there is no Wifi network configured at all.

In the first case we can probably do something, like patching wificonnect.sh, or, something like this in the startup script:

> if [ -f "/tmp/sd/wpa_supplicant.conf" ]; then
> 	kill `ps | grep wpa_supplicant | grep -v grep | awk '{print $1}'`
> 	/home/base/tools/wpa_supplicant -c/tmp/sd/wpa_supplicant.conf -g/var/run/wpa_supplicant-global -Dnl80211 -iwlan0 -B;
> fi

This should then, at every startup, kill the standard Yi wifi connect and re-start with the custom one. 

However...if the "Not paired" voice message is triggered by a Yi-config check (rather than actually not being able to connect to the previously configured Wifi) then this won't work. Because the Yi software will keep thinking that the camera is not paired/configured, even if we 'reset' the Wifi connection. This would only work if the Yi software actually checks/verifies the Wifi/IP connection upon startup (and the "not paired" voice message is based on that).

Normally if the cam loses its connection the blue light just starts flashing and when the Wifi comes back it reconnects. That does not cause the "Not paired" voice message. So I wonder what may trigger that scenario. Maybe if it does not detect the Wifi network at all while the camera is booting..or something similar like that?

roleoroleo, 2020-05-31:
I switched off my AP and rebooted the cam.
I can't ear "not paired" message.
I tried with soft reboot (reboot command in shell) or hard reboot (removing power) and the cam doesn't speak.
Then I turned the router back on and the cam reconnected automatically.
Probably the "not paired" happens when cam loses the config. But why?
I have been restarting 2 cams continuously for almost a year and I have never had this problem.

hocker96, 2020-05-31:
My issus is, I can't write nothing on camera. No text file you can edit, you can't Upgrade anything. I think the rom is explode. How can I write it new (with dd) ? 

Why I don't know. I Show as tomorow.

IT is possible to start the camera from Sd card? Thats better, because you come back if you have issus. You have enough memory.
You can patch the fstab for this I think.
You create a image file with all partions, systemroot ext4, home ext4, and a small vfat, that you expand by first start.

roleoroleo, 2020-06-01:
This cam has 2 partitions:
1. rootfs is a squashs
2. home is a jffs2

squasfs is a read only file system, so you can't write anything on the following dir: bin, etc, lib, mnt, root, sbin and usr.
jffs2 is read/write, so you can write on /home.

If you can't write on /home, you probably don't have any free space left.
Check with the command df your state:
```
root@yi-hack-allwinner:~# df
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776     10020      1756  85% /home
tmpfs                    32768        84     32684   0% /tmp
/dev/mmcblk0p1         7743488   5421532   2321956  70% /tmp/sd
```
home partition is on /dev/mtdblock3
To recover the cam if the home partition is full, log in with telnet or ssh and remove the hack deleting the folder /home/yi-hack

About starting the cam from sd, it must be supported by u-boot, I don't know if it's possible.

MrPatate87, 2020-06-19:
Have the same problem on one of my yi home camera.
Only one on two camera have this problem, when I disconnect it form power or remove the sd card, I always must remade the wifi synch from yi home app. Never have this problel with the other camera.
I double check the yi home cloud is enable in this camera.

roleoroleo, 2020-06-19:
Check the fw version (yi fw).

Xandrios, 2020-06-19:
Does the yi firmware version not get overwritten/reset when we apply the hack?

If that is not the case it would make sense to every now and then revert the hack, upgrade the yi firmware, and re-apply the hack. It thought though that applying the official fiormware updates don't make much sense because the hack-firmware would bring everything back to the version that the hack was based on originally?

MrPatate87, 2020-06-19:
> Check the fw version (yi fw).

Can't check the yi firmware version cause this webcam won't connect to yi app anymore (but i made the last update before  apply the hack, the update will have 2 mounth now.)

roleoroleo, 2020-06-20:
This hack doesn't contain files from the original yi firmware.
And it doesn't change original files **except**:

- /home/app/init.sh
- /home/base/tools/extpkg.sh

At the moment, the upgrade works with both versions:

- 8.2.0.0A_201912121707
- 8.2.0.0A_201912270941

I don't know if it could work with a future version and I think that yi wil not release a new version.

roleoroleo, 2020-06-20:
@MrPatate87 
Are you able to detect the differences between the 2 cams?

MrPatate87, 2020-06-20:
> @MrPatate87
> Are you able to detect the differences between the 2 cams?

One of the camera haven't been update before apply hack, on this one i'm on 8.2.0.0A_201912121707 yi firmware.
This one work fine with yi home app and with the hack. 
One the second one i have apply yi firmware update before apply the hack si i must be in 8.2.0.0A_201912270941 version. This camera wotk fine with the hack, but this one can't save network info and i must reconnect via yi app every time the camera is power off. More other on this camera thé wifi connection work with yi home app but after this the camera won't show in yi app.

MrPatate87, 2020-06-24:
Don't really know why but I disable the option to Recording without Cloud and reboot the camera to apply the setting, after this whenthe camera boot, i have to link my wifi with the yi home app, after that the camera will appeare in the yi home app and I can re enable Recording without Cloud option, reboot the camera and the wifi parameter was saved (so i don't need anymore to do the wifi setting with the app)
Don't relly understand why it works but it work for me.
Thanks fo @roleoroleo for all your help and for this great hack.

Schm1tz1, 2020-07-10:
Hi all,

is there any update on the original issue? Looking at /home/app/init.sh the sd card is mounted in the very beginning even before the wifi kernel module is loaded. 
Modifying /home/app/wifi.sh to use a (configurable) wpa.conf from SD card instead of /tmp/wpc should do the trick:
`echo -e "ctrl_interface=/var/run/wpa_supplicant\nap_scan=1\nnetwork={\nssid=\"TP-LINK_gzy\"\
                                                                                            
ifconfig wlan0 up
sleep 1
/home/base/tools/wpa_supplicant -c/tmp/wpa.conf  -g/var/run/wpa_supplicant-global -Dnl80211
                                                                                            
killall udhcpc
udhcpc -i wlan0 -b -s /home/app/script/default.script
                                                                                            
rm /tmp/wpa.conf`

latigreplata, 2020-09-07:
Same issue here; have to reconnect to wifi (by scanning the QR code) every time I reboot the camera. Issue affects two cameras, both on 8.2.0.0A_201912270941 firmware. Hack works flawlessly on both cameras though, it's just the whole rebooting-and-register-wifi-issue that is a bit annoying. It also means that if the power is lost somehow, the cameras wont work again until manually reconnecting.

Anything I can do to help identify this issue/bug?

roleoroleo, 2020-09-07:
Did you have the same problem before applying the hack?

rb2k, 2020-10-05:
Another data point:

I 'lost' a camera while unpugging / plugging it into a solar cell + battery pack.
All of a sudden it said "ready to pair".

(both camera on the latest firmware using the latest hack. I set them up 2 days ago.)

roleoroleo, 2020-10-06:
This is a very annoying problem, I have to find a way to solve it.
A lot of issues concern this problem.

roleoroleo, 2020-11-21:
I know this is an old issue but I made a new wiki page to change wifi credentials.

rb2k, 2020-11-21:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Change-WiFi-credentials

Awesome!

roleoroleo, 2020-11-21:
I'm not sure if this method solves the problem but you can try.

roleoroleo, 2020-12-19:
@rb2k 
Did you try the wiki page method?

andriej, 2021-01-05:
@roleoroleo I'm just trying (received 9FUS today) - doesn't work, so hack works, config is saved, MQTT connects etc. but after a reboot/powerdown it's 'waiting to connect' message played.

closed 2021-02-15 by roleoroleo