roleoroleo/yi-hack-Allwinner #52: Repair 'interrupted' mp4 video files

Xandrios, 2020-05-31:
I'm trying to find a way to make 'cut off' mp4 files playable. These are files that were copied from the camera while the Yi software was still writing to them. So far without much luck unfortunately.

The video data is there. But from what I understand the mp4 format can have some 'metadata' at the end of the file, and this would be required to play the video. 

There are some tools and services that theoretically can do this for you. I've tried these:

* [recover_mp4](https://www.videohelp.com/software/recover-mp4-to-h264) Which requires the unfinished file and a working 'complete' file as input.  (Windows)
* [untrunc ](https://github.com/ponchio/untrunc)Same two-files for input (Unix)

And various online recovery services. None of these seem to work...I didn't get a working video files from any of these. Most likely some crucial mp4 parameters, specific to this camera/model, need to be used in this repair process. 

Has anybody tried this, or already know of another way to repair interrupted mp4 video from the Yi home cam?

SecAficionado, 2020-06-25:
I'm guessing you did, but just to make sure: did you try **ffmpeg**? It might be able to read the beginning of the file and start working on it, and you tell it to stop just before the moment of interruption.

It would really help if you have a sample file to run tests.

Giuserver, 2022-02-24:
> I'm trying to find a way to make 'cut off' mp4 files playable. These are files that were copied from the camera while the Yi software was still writing to them. So far without much luck unfortunately.
> 
> The video data is there. But from what I understand the mp4 format can have some 'metadata' at the end of the file, and this would be required to play the video.
> 
> There are some tools and services that theoretically can do this for you. I've tried these:
> 
> * [recover_mp4](https://www.videohelp.com/software/recover-mp4-to-h264) Which requires the unfinished file and a working 'complete' file as input.  (Windows)
> * [untrunc ](https://github.com/ponchio/untrunc)Same two-files for input (Unix)
> * [Grau mp4 repair](http://grauonline.de/cms2/?page_id=5) (Windows)
> * [Stellar repair for video](https://www.stellarinfo.com/disk-recovery/video-repair.php) (Windows)
> 
> And various online recovery services. None of these seem to work...I didn't get a working video files from any of these. Most likely some crucial mp4 parameters, specific to this camera/model, need to be used in this repair process.
> 
> Has anybody tried this, or already know of another way to repair interrupted mp4 video from the Yi home cam?

Hi!
You solved?

closed 2024-05-14 by github-actions[bot]