roleoroleo/yi-hack-Allwinner #55: RTSP - restored h264grabber -> rRTSPServer chain

roleoroleo, 2020-06-17:
After identifying a series of stability problems on the new version of rRTSPServer (especially on the MStar platform) I restored the old method with h264grabber.
This change is a breaking change regarding the ports of the RTSP service: low and high resolution streams cannot share the same port because they are 2 running processes (as in version 0.1.3).
If anyone wants to try it ...

This is the changelog:

- Add rtsp and snapshot links to the homepage
- Restored chain h264grabber -> rRTSPServer from 0.1.3 (with and withoud audio)
- Bump mqttv4 to the last version (move mq and add retains)
- Update live555 to the last version
- Add ipc_multiplexer and ipc_read (if you want to develop your own events manager)
- Add patch to unblock open action on audio fifo
- Fix video cleaning procedure
- Move rmm start position
- Add reset to factory feature
- Add scripts to cleanup sources
- Change output file option in imggrabber
- Fix bug about swith on/off camera

[y20gam_0.1.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4792042/y20gam_0.1.6.tar.gz)

digitalchou, 2020-06-19:
Ill try it
Need I to restore original sytem,or simply write that to a sd card and boot up the cam?

roleoroleo, 2020-06-19:
Simply copy to sd and restart.
Backup your config.

pbanj, 2020-06-20:
seems to have a lot of lag. like a good 30 seconds or more

murigny, 2020-06-20:
now, can connect on high res stream but lost signal every 2 minutes or have some glitch.

digitalchou, 2020-06-20:
Lot of lag here so, but dont crash. 
Release version 0.1.6 its neerly perfect, but it have cuts every time there is heavy movement on image.
I have an IOS app that have an option to force TCP conecction, and with this app I have no cuts ever.
Is there anyway to modifi version 0.1.6 to reject udp connections?
Any mod on startup script or so. It would be a simply workaround for my use case. 
When I use that app forcing TCP  the image timing is perfect on any situation, and i have no cuts, but in OBS the cam runs well until a heavy movement on screen (high bitrate I supposse) make the stream crash and reconnect on few seconds, but it make the cam unnusable.
Moding source code and compiling is a hard way to me, but I can tray if theres no other way.
Any Idea ?
I tried the sysctrl on ssh but busybox lacks of sysctrl.
Im a nebbie.
Thanks in advance.

murigny, 2020-06-22:
Hello, i will try to build with a lower framerate (15.0) on rRTSPServer.patch ...

roleoroleo, 2020-06-22:
> Lot of lag here so, but dont crash.
> Release version 0.1.6 its neerly perfect, but it have cuts every time there is heavy movement on image.
> I have an IOS app that have an option to force TCP [...]

Many users use udp, I can't remove it.
And I don't know if I able to edit the live555 code.
I think it's better to force the client to use tcp if it works better in your system.

pbanj, 2020-06-22:
maybe add an option to force tcp or udp

minax007, 2020-06-26:
I have just tested the firmware and I really like it. 
It is indeed a bit laggy - but in my system setup (only low stream - nothing else) it is acceptable. 
The lag is about 5 seconds. 
The firmware runs stable. 

The last weeks I was running 0.1.3 because the newer versions were not stable for me. 

What I really like with the new firmware is the link to the RTSP-stream. So I do not have to look in the documentation each time, and I do not have to modify it with the correct IP address - because it is already mentioned in the link. 

Thanks a lot roleoroleo for this!

In case you are interested in my wish list: 
I would like to see the RTSP-stream as a link to a new page in the browser. My browser is set up in that way that it does automatically start VLC when I try to open an RTSP link. But the actual way, by just copying and pasting is also fine for me. 
And I think you had mentioned somewhere that it is not possible to have different resolutions than "low" and "high". If you find a way to select other resolutions that would be really great. 

Nevertheless in total so far I fully appreciate your excellent work. 
Thanks!

roleoroleo, 2020-06-27:
> In case you are interested in my wish list:
> I would like to see the RTSP-stream as a link to a new page in the browser. My browser is set up in that way that it does automatically start VLC when I try to open an RTSP link. But the actual way, by just copying and pasting is also fine for me.

No problem, it's a simple href, I will add it to the next release.

> And I think you had mentioned somewhere that it is not possible to have different resolutions than "low" and "high". If you find a way to select other resolutions that would be really great.

Sorry but this is not possible (for my knowledge).

> Nevertheless in total so far I fully appreciate your excellent work.

Thank you.

closed 2020-07-14 by roleoroleo