roleoroleo/yi-hack-Allwinner #56: Yi iot 1080p outdoor

fernandodpr, 2020-06-21:
Firts of all It's my fisrst time publishing on github so sorry If I'm not in the correct tab. 
A few weeks ago received an Yi IOT camera that I read that haves the Allwinner  chipset. [Link to the product ](https://es.aliexpress.com/item/4000120325499.html?spm=a2g0s.9042311.0.0.274263c0N5u4wV). 

I would like to work to make the camera compatible with the wi-hack firmware. 

Firmware version: 7.1.00.19a_201910181012
I found a github [page](https://github.com/rage2dev/YiOutdoor) made by @rage2dev that explains how to load a custom firmware. 

So, I offer myself to test the installation in my camera. Should I install the firmware renaming the .tgz file? Do you think that more adjusting would be necesary?

pbanj, 2020-06-22:
Prob best to open it up and see what soc it actually has

roleoroleo, 2020-06-22:
If you have telnet access, try

cat /proc/cpu

matteopiccioni, 2020-06-24:
I'd like to have Yi1080p outdoor support
Your firmware has mqtt and webservices support that are great features
TheCript0 fw has not mqtt and webservices.
Thanks

fernandodpr, 2020-06-25:
Finally I managed to continue the investigation.  I've just uploaded some photos of the inside, can't find anything related to the realtek chipset googling. 

The output of _cat /proc/cpu_ is:

```
system type             : RLX Linux for IPCam Platform
machine                 : Unknown
processor               : 0
cpu model               : Taroko V0.2  FPU V0.1
BogoMIPS                : 497.66
tlb_entries             : 64
mips16 implemented      : yes
```

[image]
[image]
[image]

roleoroleo, 2020-06-25:
Sorry but this platform is not supported.
This cam needs another hack.

closed 2020-07-14 by roleoroleo