roleoroleo/yi-hack-Allwinner #58: Any chance to set a SSH root pwd post-boot?

Catfriend1, 2020-06-23:
Hi,

is there any chance to setup a SSH password post-boot as I've already seen the /etc/passwd is on a read-only partition. That causes "passwd root" to fail.

```
/dev/root on / type squashfs (ro,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=30528k,nr_inodes=7632,mode=755)
proc on /proc type proc (rw,nosuid,nodev,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,relatime)
/dev/pts on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
/dev/mtdblock3 on /home type jffs2 (rw,relatime)
none on /sys/kernel/debug type debugfs (rw,relatime)
none on /dev/mqueue type mqueue (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime,size=32768k)
/dev/mmcblk0p1 on /tmp/sd type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
```

roleoroleo, 2020-06-23:
No, at the moment.
We should change the rootfs type (jffs2) but it's a risky operation.

Catfriend1, 2020-06-25:
I've put this code in the startup script, and it successfully set-up my root password for SSH.

```
ROOT_PWD="$(cat "/home/yi-hack/etc/system.conf" | grep "^PASSWORD=" | cut -d '=' -f 2-)"
ROOT_PWD_MD5="$(echo "${ROOT_PWD}" | mkpasswd --method=MD5 --stdin)"

cp -f "/etc/passwd" "/home/yi-hack/etc/passwd"
sed -i 's|^root::|root:*:|g' "/home/yi-hack/etc/passwd"
mount --bind "/home/yi-hack/etc/passwd" "/etc/passwd"

cp -f "/etc/shadow" "/home/yi-hack/etc/shadow"
sed -i 's|^root::|root:'${ROOT_PWD_MD5}':|g' "/home/yi-hack/etc/shadow"
mount --bind "/home/yi-hack/etc/shadow" "/etc/shadow"
```

commit 0b2f79db (2020-06-25): Set SSH root password (fixes #58)

Purpose:
- This sets the SSH root password for better security.
- It is also used for PureFTPD FTP access with the root / {password} credentials.

Verified working on my Yi Home 1080p v3 camera.
commit fe2306ad (2020-06-25): Merge pull request #64 from Catfriend1/ssh_root_pwd

Set SSH root password (fixes #58)
closed 2020-06-25 by roleoroleo