roleoroleo/yi-hack-Allwinner #59: Can’t flash on 9FUSY31MF0

andriy7811, 2020-06-23:
Hi all, just bought a YI 1080 home on Amazon ( with Firmware 8.2.0.0A_201912270941 ). 

With the 16GB SD formatted FAT32 I copied the file + 2 folders, switched off, inserted SD and switched on.

Yellow light flashes for few seconds then it becomes blue and flashing. Waited 30+ minutes and nothing happen.
In this condition the camera is not connected in my network. As soon as I remove the SD the Camera connect to the network with its original firmware.

Any clue?

Thanks a lot

roleoroleo, 2020-06-23:
Remove all files from sd and try again.
If doesn't work, try another card.

andriy7811, 2020-06-23:
Thanks for your reply, I have tried with 2 different 16GB SdCards: Transcend and SanDisk and one try also with a 32GB SanDisk.

After having them formatted to FAT32 I copied these files:

Home_y20gam.stage
Factory/config.sh
Factory/factory_test.sh
Newhome/base/tools/extpkg.sh

What files do you suggest to remove?

I get same behavior with both SD Card ( both brand new ) so I thought it could be linked to a more recent model ( on the box it is reported Date: 2020.03 )

I have tried even earlier versions ( I read somewhere that flashing an earlier version and then upgrade via web interface has helped in some cases.. ) but there is no way unfortunately

roleoroleo, 2020-06-23:
The firmware version 8.2.0* means that the cam is compatible with the hack.

Are you sure the fw is not hacked?
How did you check it?

andriy7811, 2020-06-23:
To be honest I’m no 100% sure:

I’m just sticking on these 2 behaviors:
- while flashing the firmware after some yellow led flashing it switch tu blue led flashing and it won’t change its state ( no steady light ), so I am assuming the process did not complete
- Right now I am trying to reach the web interface to its address:8080 but it’s not reachable ( while the device is listed in my router and the camera is available in its native app )

Any other thing I could do to check?

roleoroleo, 2020-06-23:
Web page is enough.

jjansen85, 2020-06-27:
Same problem on 9FUSY310OZ.
Then tried to install 0.1.3 from SD-card. Worked very well (only a few minutes).
Then did a update via web page to 0.1.6, which succeeded.

roleoroleo, 2020-06-27:
The only difference between manual upgrade and online upgrade is that the first one doesn't backup/restore the config.
Both methods use the files on sd.
Probably an error copying the files.

andriy7811, 2020-06-29:
Tried to do the same with 0.1.3 but this approach did not work on my end. Endless blue led flashing condition.
Still trying to find out a solution.

roleoroleo, 2020-06-29:
Try this procedure:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Dump-your-backup-firmware-(SD-card)
to check if the startup hack works.

andriy7811, 2020-06-29:
I believe it doesn’t work.
In the sd root I only find Factory folder I extracted with included config.sh and factory_test.sh files

Tried with 2 different SD cards.  

I also thought my cam was faulty in the SD reader but if I include SD cards in normal mode the cam records motion triggered videos properly and I can see them on my pc.

I’m used to these kind of actions, I have a quite well integrate Home Assistant system .. so really I cannot understand why I can’t succeed with this...

roleoroleo, 2020-06-29:
Could you take a picture of your sd file system?
A Windows explorer is enough.

andriy7811, 2020-06-29:
Here it is

https://ibb.co/5cjLTqK

roleoroleo, 2020-06-29:
Is thie the hack or the backup procedure?

andriy7811, 2020-06-29:
Yes this is what I extracted from backup.tar.gz

andriy7811, 2020-06-29:
Backup procedure

roleoroleo, 2020-06-29:
I don't know.
The only way to understand the problem is connect a serial interface.
But it's complicated.

andriy7811, 2020-06-29:
I have a SH-U09 dongle, would it work for the scope? Any guide to follow to try to give some info?

roleoroleo, 2020-06-29:
Yes it works.
See the wiki.

jonferreira, 2020-07-06:
@andriy7811 any joy? 

I seem to be having a similar issue. As soon as I add the Factory folder the camera get unavailable even though it seems to boot fine (steady blue LED)

```
D:\>dir
 Volume in drive D has no label.
 Volume Serial Number is 0000-00C0

 Directory of D:\

06/07/2020  16:51    <DIR>          record
06/07/2020  16:46    <DIR>          log
07/05/2020  16:03    <DIR>          Factory
               0 File(s)              0 bytes
               3 Dir(s)  15.912.910.848 bytes free

D:\>cd Factory

D:\Factory>dir
 Volume in drive D has no label.
 Volume Serial Number is 0000-00C0

 Directory of D:\Factory

06/07/2020  17:52    <DIR>          .
06/07/2020  17:52    <DIR>          ..
07/05/2020  16:03               638 config.sh
03/05/2020  16:31                 0 factory_test.sh
               2 File(s)            638 bytes
               2 Dir(s)  15.912.910.848 bytes free

```

andriy7811, 2020-07-13:
I had no luck with serial interface. 
I gave up. Looking for a different camera to integrate in Home assistant...

roleoroleo, 2020-07-14:
Problems soldering the interface?

closed 2020-11-21 by roleoroleo