roleoroleo/yi-hack-Allwinner #60: Can.t change password

Czapla-dev, 2020-06-24:
When he tries to change the password using ssh he gets this message:
`root@yi-6fe1:~# passwd
Changing password for root
New password: _password_
Retype password: _password_
passwd: /etc/passwd: Read-only file system
passwd: can't update password file /etc/passwd
`

murigny, 2020-06-24:
you can't change root password, it's on read only partition of yi firmware.

Czapla-dev, 2020-06-24:
In this case, how do you change the password? I'm trying to follow the instructions at Home Assistant page.
https://www.home-assistant.io/integrations/yi/#changing-the-ftp-password

murigny, 2020-06-24:
You can enable username/password by using yi-hack web page. It will activate authorisations for ftp,ssh,telnet,rtsp and http ( same for all )

roleoroleo, 2020-06-24:
The username/password setting works only for http, rtsp and onvif.
ssh and telnet use passwd but rootfs is read only, so you can't change it.

roleoroleo, 2020-06-24:
Instructions at ha site are for an older version/hack.

Czapla-dev, 2020-06-24:
I tried without changing the password, but ftp has no video, the image via rtsp but the preview does not work ...

roleoroleo, 2020-06-24:
Don't use yi integration on ha. It's for an older version.
See the wiki.

Catfriend1, 2020-07-12:
Meanwhilst, a PR has been merged which allows to set the ssh password from the web ui.

closed 2020-11-21 by roleoroleo