roleoroleo/yi-hack-Allwinner #61: MQTT - Motion Start and Stop messages

matteopiccioni, 2020-06-24:
Hello,
in MQTT page I am trying to configure 'Motion Start' message in this way:
`{"ipcam":"sala_on"}`
but when I save and reload the page, the 'Motion Start' text is only: sala_on

Using ssh and msquitto_pub I was able to send messages in this way:
`mosquitto_pub -h 192.x.x.x -u mqtt_usr -P mqtt_pwd -t home-assistant/ingresso/motion_detection -m '{"ipcam":"sala_on"}'
`

This is my sensor configuration:
binary_sensor:
```
  - platform: mqtt
    name: Movimento Sala
    payload_on: "on"
    payload_off: "off"
    device_class: motion
    state_topic: "home-assistant/ingresso/motion_detection"
    value_template: >
      {% if value_json.ipcam == 'sala_on' %}
        on
      {% elif value_json.ipcam == 'sala_off' %}
        off
      {% else %}
        {{ states('binary_sensor.movimento_sala') }}
      {% endif %}
```

In witch way should I configure the 'Motion Start' message in more complex way (json data format)?

Thanks!

roleoroleo, 2020-06-24:
Why do you need json payload?

matteopiccioni, 2020-06-24:
Well ... I don't necessarily need structured JSON to do what I need ...
It was more of a didactic purpose, to better understand how things work :)

roleoroleo, 2020-06-24:
There is a problem with the string parser.
When you save the json payload from the web page, the file .conf contains only "sala_on".

matteopiccioni, 2020-06-24:
Thanks roleoroleo, at the end I use a simple string 👍 
I think you can close

ilmisu, 2020-06-26:
May I continue this topic? I have my yicam 1080 which sends presence detection payloads as well if everything is still in the room.
Any suggestion to check or adjust?

roleoroleo, 2020-06-26:
Could you explain me better your purpose?

ilmisu, 2020-06-26:
sure, I'd like to use it as an alert if I'm not at home, BUT I show you the record of last night...
[image]
as you can see, nobody was in the kitchen at 3AM, but presence is continuously detected...

roleoroleo, 2020-06-26:
This problem was already discussed for MStar based cams.
There isn't a final solution, but this is my suggerstion:

- Check the electronic devices in the room (I remember a lot of detection when my phone receveid wa messages and the led flashed)
- Set the sensitivity to low

ilmisu, 2020-06-26:
the sensitivity is already set to low.
electronic devices, uh? I'll check..
but, does it use the same alert of the stock fw?

roleoroleo, 2020-06-26:
> the sensitivity is already set to low.
> electronic devices, uh? I'll check..
> but, does it use the same alert of the stock fw?

Yes, the motion detection is "sniffed" form the stock fw.
The motion algorithm is developed by yi.

closed 2020-08-29 by roleoroleo