roleoroleo/yi-hack-Allwinner #63: Is it possible to install hack on brand-new camera?

SecAficionado, 2020-06-25:
Hi, I am trying to setup a brand new camera, but I do not want to use the Yi app at all. I'd rather not create a user id, and I'd rather not use their cloud service at all. I want to take a brand-new camera and connect it to Home Assistant.

On issue #49 @roleoroleo mentions that is is possible to use a wpa_supplicant.conf file to set the WiFi connection parameters, but there are no details. I think that using that hack, it might be possible to flash a new camera and have it connect to the network.

Could you please say whether it is possible to apply the hack to a camera without using the app/cloud, and then configure it? If it is possible, could you please explain the necessary steps to get there?

I am just now getting familiar with everything here, but I don't mind poking around and working out the details, if you can point me in the right direction.

Thank you in advance!

SecAficionado

roleoroleo, 2020-06-25:
I think nobody has tried this method.
I don't know if it's works.

Probably it's easier to use a temporary email without personal data for the registration process (and save it if you need it again).

pbanj, 2020-06-25:
I used a temp email to make an account for the app, then uninstall the app after I install the cfw

SecAficionado, 2020-06-25:
Thank you, both. I'll do that.

closed 2020-06-25 by SecAficionado