roleoroleo/yi-hack-Allwinner #65: Fyi: IP Cam Viewer App support is coming soon

Catfriend1, 2020-06-25:
Hi,

I've contacted the author of the Android app "IP Cam Viewer Lite/Pro", Robert Chou, to add support for the Yi Hack to the app. We've exchanged infos and finally I got a beta build which worked fine from him :-).

It can:
- View snapshot "stream" (ch9)
- View sd/hd stream(ch0, ch1)
- Turn motion detection (cloudless with microsd) on/off
- Reboot the camera

I hope it will arrive on the store soon.
Link: https://play.google.com/store/apps/details?id=com.rcreations.ipcamviewer

Kind regards
Catfriend1

murigny, 2020-06-25:
lol

it's a joke pub
All rtsp apps can support this hack.

Catfriend1, 2020-06-25:
I don't think so because it can control motion detection on/off which I personally use very often.

Catfriend1, 2020-06-28:
Update arrived.

closed 2020-06-28 by Catfriend1

Merlynda, 2020-07-01:
Can you add support for hack to this app?

https://channelstore.roku.com/details/143683/ip-camera-viewer-pro

Catfriend1, 2020-07-01:
Please contact the app author, its just what I did :)