roleoroleo/yi-hack-Allwinner #66: How to reset camera

zatarra, 2020-06-25:
Can anyone tell me how to reset the camera?

It looks like a weird question given that no one else asked the same thing (therefore I assume the answer should be sitting right in front of my eyes), but I'm having a hard time resetting it.

I've configured the camera, but now I need to move to another place that has a different wireless network. How can I reset it? 

I tried to press the reset button with and without the sd card in it. I tried to enable and disable cloud (and repeat the same procedure) and I also tried to find a factory reset binary.

Do I need to flash it with the original firmware? 

Firmware Version | 0.1.6
Base Version | 8.2.0.0A_201912270941
Model Suffix | y20gam   (9FUS) Yi Home 1080p

roleoroleo, 2020-06-25:
You don't need the original fw because the original fw is under the hack.
Switch off the button "Disable cloud" and reset the cam as if it were original.
Repeat the pairing with the app and reconfigure it normally.
At the end, switch on again "Disable cloud", if you want.

zatarra, 2020-06-25:
You are absolutely right. I think the issue here is that I was not pressing long enough. It worked just like you said!

closed 2020-06-25 by zatarra