roleoroleo/yi-hack-Allwinner #67: Mount network storage in /etc/fstab

pniewiadowski, 2020-06-27:
Hi There,

first of all, thank you for your awesome work!

This is more like a question. Do you know if we can mount network storage (NFS, Samba) using fstab configuration on yi cam?
I was trying to do this but the fstab seems to be read-only protected so I wasn't able to do this.

My goal is to write video data directly on my QNAP NAS, instead on SD card.

roleoroleo, 2020-06-27:
Nfs and samba are not supported at kernel level.
-------- Messaggio originale --------Da: pniewiadowski <notifications@github.com> Data: 27/06/20  17:51  (GMT+01:00) A: roleoroleo/yi-hack-Allwinner <yi-hack-Allwinner@noreply.github.com> Cc: Subscribed <subscribed@noreply.github.com> Oggetto: [roleoroleo/yi-hack-Allwinner] Mount network storage in /etc/fstab
  (#67) 
Hi There,
first of all, thank you for your awesome work!
This is more like a question. Do you know if we can mount network storage (NFS, Samba) using fstab configuration on yi cam?
I was trying to do this but the fstab seems to be read-only protected so I wasn't able to do this.
My goal is to write video data directly on my QNAP NAS, instead on SD card.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/roleoroleo/yi-hack-Allwinner/issues/67",
"url": "https://github.com/roleoroleo/yi-hack-Allwinner/issues/67",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

pniewiadowski, 2020-06-27:
What about trying to connect from within yi camera to an external FTP and send video files there?

roleoroleo, 2020-06-27:
These methods are possible:

- ftp to and from an external server
- scp/sftp to and from an external server
- gdrive sync (search in  MStar issues)
- web services from an external server (see wiki)

But you have to write your own script.

closed 2020-06-27 by pniewiadowski

Catfriend1, 2020-06-28:
Writing to ftp works nicely for me, see my script at https://github.com/roleoroleo/yi-hack-Allwinner/pull/68 .
I'm handing the records over to my debian vm and this is doing the following things to filter videos out I don't want to get via telegram push in the end because they contain car, tree etc. "fake" motion.

- scan with dvr-scan , if no motion delete instead of sending to telegram
- get duration of the video via "mediainfo" , if duration ==59 delete and skip the video.

If nothing above, send the video file via telegram push to my phone. This way , I only get "true" alarms, e.g. when a human walks into my garden.