roleoroleo/yi-hack-Allwinner #69: How to completely uninstall Yi-Hack the right way?

Catfriend1, 2020-06-29:
Hi,

I'd like to remove YiHack completely and revert the firmware of the camera to stock. What steps do I have to take and which could be possible pitfalls in regard to bricking the camera accidentially?

Thank you.

Kind regards
Catfriend1

roleoroleo, 2020-06-29:
To remove the hack you have to delete the /home/yi-hack directory and restore 2 files of the original firmware:
- /home/app/init.sh
- /home/base/tools/extpkg.sh

If you make a mistake editing the last two files the cams could brick.
I could create a "sd-unhack" but I need time.

Catfriend1, 2020-06-29:
Ok, reading through the yi hack scripts, I expect this could work:

- /home/app/init.sh

Replace

```
sleep 2
./rmm &
sleep 2
#./mp4record &
#./cloud &
#./p2p_tnp &
#./oss &
#./watch_process &
/usr/sbin/telnetd &
/home/yi-hack/script/system.sh
```

by 

```
sleep 2
./rmm &
sleep 2
./mp4record &
./cloud &
./p2p_tnp &
./oss &
./watch_process &
```

- /home/base/tools/extpkg.sh

Should be okay to leave as it is (I don't have an unhacked original).

roleoroleo, 2020-06-29:
Here it is:
[original.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4848278/original.tar.gz)

Remove /home/yi-hack and the symbolic link /home/yi-hack-v4

Catfriend1, 2020-06-29:
Cool, thanks :)

roleoroleo, 2020-07-01:
@Catfriend1 
I forgot a detail.
There is another file "modified".
You have to rename /home/app/cloudAPI_real ---> /home/app/cloudAPI
overwriting the hacked /home/app/cloudAPI.

Catfriend1, 2020-07-02:
Ah yes , I noticed it beside reading the scripts. It's the cloud on/off :).

roleoroleo, 2020-07-02:
Probably it's a good idea to create a new "sd-unhack".

roleoroleo, 2020-07-02:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack

Catfriend1, 2020-07-02:
Great work, thanks :-)

closed 2020-07-02 by Catfriend1