roleoroleo/yi-hack-Allwinner #70: 9FUS model, 8.2 firmware, and a QG2101A chip. after 13 straight days, can't get to work.

IssisX, 2020-06-29:
Hey everyone. First off, I'd like to say thank you to the one's working so hard on all of this, especially @roleoroleo, as your work has been absolutely crucial to so many of us noobs, especially to those of us who are attempting to learn this stuff, to then subsequently contribute to these Yi camera projects.

Ok, now for my issues:

@roleoroleo, hey man, it's nice to finally [attempt to] speak to you. I've been lurking around here for quite some time, keeping tabs on your and others' work and releases for my yi_home_1080p, 9FUS model, 8.2 firmware, and QG2101A chip. This is my first post ever on Git, and there's a good reason for it.

So, I have been on this problem for ~10 hours a day for 13 straight days now, with effin *BLEEP* to show for it besides a brand new, perhaps bricked, camera.
 offer me.,
Initially, I wasn't privy to the info that I am now, and accidentally loaded your yi-hack-v4 onto my yi_home_1080p, 9FUS model, 8.2 firmware, and QG2101A chipped camera, and also have yet to be able to successfully fully compile the correct version for my cam, your Allwinner version, and am now here, on my knees, asking desperately for any help that you or any other similarly qualified person can offer. 

if it's possible for someone to send me files and the exact file/folder structure that I need to put on the sd card, I would greatly appreciate it. I'm just unsure as to what needs to be done as it's all just running together in my head at this point like mush.

Also, I do have a couple of USB UART serial dongles but have not been able to correctly attach/solder the wires to the TX, RX, and/or GND pads on the chip because my soldering iron is just too large for this minuscule job. 

I can surely show/send you anything you might need to help figure this out upon request, i.e. the logs written when you reset the cam with the SD card inside, or anything you may instruct me to do. 

For context, this is what my cam does when I put my SD card in with your Allwinner hack loaded. As soon as the usb power cord is connected, the light turns orange for a second or two, then switches to flashing blue for maybe 4-5 seconds, then back to flashing orange for maybe 20-30 seconds, then, it just boot-loops this process over and over. I think it's because I've been unable to correctly compile or find the extracted firmware needed to build the hack, and perhaps the cam is looking for it and not finding it. I don't know, I'm sure i sound stupid, but hey, oh well. 

So, please roleoroleo, git community, I'm in desperate need of your expertise', and am willing to do anything to assist in this. Thank you very kindly, and I hope to hear from you soon.

IssisX, 2020-06-30:
Ok, so I have been able to compile the files I need myself, (yui-compressor and a couple of others were the issue) but still, the same boot looping process. And, when the SD card is out, it flashes yellow for about 30 seconds, and boot loops that process 

When I have the correct files on the SD card, it will flash yellow for a few seconds, and then flash blue for maybe 4 -5 seconds, and then switch to a stale yellow light for significantly longer than with any other configuration I've tried, so I think it's actually trying to get going.

I think if I had a script I could run, that would allow me to set the DIR paths correctly n such, that I could perhaps get it going. I dunno. Just an idea, because I think when I tried a few others' hacks that were not for my camera, I probably messed it where it wants to look for its boot

roleoroleo, 2020-06-30:
Let's take baby steps.
Is the cam working with the original yi app? Or not, after installing yi-hack-v4?
If it doesn't work you need to restore the original fw and I don't know if it's possible.

IssisX, 2020-06-30:
Thanks for replying bud, and for sure, good idea. Btw roleoroleo, for more context and better communication between the two of us; while I am a n00b in this situation, I've been coding n scripting n such in several different languages and more for over 20 years, just FYI brother, to help smooth over and speed up any rapport we may be building. I'm already beginning to catch on in this area, and am sure I will be able to contribute something like a debricker if someone else doesn't beat me to it.

Now, the cam worked as it should have with yi app, and now it does not after installing v4. I believe it's looking for files that now, because I installed the wrong version, arent in the correct spot...could be wrong of course. Also, a big thanks once again, I super appreciate your time man.

roleoroleo, 2020-06-30:
Try to use the backup procedure described here: https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Dump-your-backup-firmware-(SD-card)
If it works, we can access to the cam and try to restore the corrupted file system.

IssisX, 2020-06-30:
I have actually tried that method, but I will try it again in a sec. If you know of a name of a script that runs on the cam upon booting, I'm thinking that's key to this mess

IssisX, 2020-06-30:
Alright, got the backups done brother. :). What I did was rename it to equip_test.sh and placed it in a folder named "test" on root of sd. One of the wrong hacks I tried is the reason this worked. Now, do I use binwalk or Jefferson, or upload them here?

IssisX, 2020-06-30:
@roleoroleo, here are the backup files you requested. https://drive.google.com/drive/folders/1tHTqDSwbBZyv5a9VkiWHB3OZEx5EEfUw?usp=sharing

roleoroleo, 2020-07-01:
In your home file system there is a "wonderful" mix of hacks but the cam is hacked and it should work.
If you want to clean the v4 hack, enter with telnet (it should be running) and do the following:
```
rm /home/app/busybox
rm /home/app/rtsp2301
rm /home/app/rtsp2301_watch.sh
```

IssisX, 2020-07-01:
LMFAO @"wonderful" mix of hacks. I do apologize, won't soon make the same mistake. 

But @roleoroleo, my cam doesn't stay on with SD card in or out long enough to telnet in as it's stuck in a boot loop. Will removing those files fix that? I can do it a different way if this is the case. Thanks again buddy

IssisX, 2020-07-02:
I forgot to mention @roleoroleo, that I've only been working with linux for ~1.5 years, so lots and lots to learn, and so any help you can offer, in terms of a few hints I mean, would be awesome :). I'm a Windows convert.... Well Windows, especially 10, has it's place for sure. And, with PowerShell and PStools, it's getting muuuch better. But, my gawd linux is heaven for any coder or scripter or alike!! Cant beat it.

roleoroleo, 2020-07-02:
I will prepare a special SD card to try to restore your cam.
Probably you file system is full.

Merlynda, 2020-07-02:
Damn IssisX - I hope you donated to Roleoroleo!

Gentlemen, if nothing works I will buy IssisX a new replace camera to start over with.

roleoroleo, 2020-07-02:
Try this:
[unhack.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4865368/unhack.tar.gz)

But use at your own risk.
I didn' tested it.

IssisX, 2020-07-02:
Lmao @Merlynda....wouldn't that be the bomb dot com, haha, and you're right about donating! Roleo, please point me to your donate link/button, I can't find it, and you certainly deserve it.

Now @roleoroleo, super cool of you to do this bro, thank you so much, and if you remember from the SD backup scripts you made, I had to rename them to something else because it wouldn't read Factory/config.sh, and I'm having the same issue here. The obsessive part of me isn't going to let this sit, and there are/will-be others that need/will-need this, so I'm going to try a few things to get this working.

When I tried it though, the boot process/light situation was actually different this time, and took longer to do, so something is happening but it didn't rename any of the files when it was done, so, kinda odd really.

roleoroleo, 2020-07-03:
> Now @roleoroleo, super cool of you to do this bro, thank you so much, and if you remember from the SD backup scripts you made, I had to rename them to something else because it wouldn't read [...]

This is extracted from your dump:
```
if [ -f "/tmp/sd/Factory/factory_test.sh" ]; then
	/tmp/sd/Factory/config.sh
	exit
fi
```

The file I prepared must work.

> 
> When I tried it though, the boot process/light situation was actually different this time, and took longer to do, so something is happening but it didn't rename any of the files when it was done, so, kinda odd really.

The unhack procedure remove only the hack.
It doesn't reinstall.
After running unhack, try to use the original app and check if it works.

IssisX, 2020-07-03:
I wish it did work for me roleo, but it didn't, and neither does the app after it "finishes" (which it never does because of boot-looping). I am sorry to be such a pain in the arse, truly, It's just it's the boot-looping that is the problem. It won't stay on long enough to get anything done. I thought of removing "reboot" from the end of your scipt, but wasn't sure if that was a good idea or not.

One thing that I DID manage to capture, is the camera on my network for literally 1 or 2 seconds via netdiscover right before it boot-looped again. So, that is the light in the tunnel that's keeping me from launching it in the sky on one of my 4th of July's giant bottle rockets!! Lol...

roleoroleo, 2020-07-03:
Did you try a reset to default with the pin hole, after applying unhack?

IssisX, 2020-07-03:
And also btw, I have a new Yi camera coming, should be the same model and firmware and everything. I didn't take @Merlynda to be serious about sending me one :laughing:

roleoroleo, 2020-07-03:
> Yes I did. Hey btw roleo, do you accept bitcoin?

- Try to apply this [restore.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/4870812/restore.tar.gz)
- You must copy it to a SD and power on the cam like a normal upgrade.
- The cam should reboot only one time.
- Remove the SD.
- Dump again a backup of your cam using the other archive in the wiki.
- Share it with me.

> The donate button won't work for USA ppl for some reason
If you can't donate with paypal don't worry. No problem.

chl3b, 2020-11-16:
Hi roleoroleo. I have similar issue but cant sort this out. My camera cant connect to.Wi-Fi repeating wifi password error of course all details are correct , checked with few other routers. Tried few hack to solve this but no joy, from this thread as well. Camera was hacked and was working for some period of time and suddenly stopped working with yi app which is crucial for my spouse to use simply tool. It is not my first yi cam and applying hack again was helpful. I think problem started when new fw was available for this model 9FUSY31UZZ. 1080 home.

roleoroleo, 2020-11-16:
> I think problem started when new fw was available for this model 9FUSY31UZZ. 1080 home.

Which version?

chl3b, 2020-11-17:
> > I think problem started when new fw was available for this model 9FUSY31UZZ. 1080 home.
> 
> Which version?
> 

unfortunately I didnt take a screenshot of fw page

roleoroleo, 2020-11-17:
Try to remove the hack (see wiki pages) and check if it starts.

chl3b, 2020-11-17:
> Try to remove the hack (see wiki pages) and check if it starts.
> 

I did but not helped. I ve saved backup using procedure described above here is a link
[]https://app.box.com/s/x5t0cxugqa8595cac6t5sjsclrm5l4og()

Could you take a look please.

roleoroleo, 2020-11-17:
Your cam is only half hacked.
Startup scripts are hacked but you miss whole yi-hack directory.
Try to apply hack again.

chl3b, 2020-11-18:
> Your cam is only half hacked.
> Startup scripts are hacked but you miss whole yi-hack directory.
> Try to apply hack again.
> 

YI hack is for this model or allwinner? Want to check before running.

roleoroleo, 2020-12-27:
To answer to your question, I need the yi firmware version.

risturiz, 2020-12-30:
> Hi roleoroleo. I have similar issue but cant sort this out. My camera cant connect to.Wi-Fi repeating wifi password error of course all details are correct , checked with few other routers. Tried [...]

I think we have the same camera "9FUS" "Y31"... I didn't upgrade the firmware and the hack **y20ga** works... My actual firmware is 8.2.0.0A_201912270941 and update is 8.2.0.0A_202012041528... So maybe there is a problem with firmware upgrade?

roleoroleo, 2020-12-30:
Maybe...
I will download the new version and check it.

roleoroleo, 2020-12-30:
Tested with 8.2.0.0A_202012041528.
It works correctly.

risturiz, 2020-12-31:
> Tested with 8.2.0.0A_202012041528.
> It works correctly.

Btw, you have to remove the hack to upgrade the firmware? Mine don't upgrade at all :-)

roleoroleo, 2020-12-31:
Yes, it's recommended.

risturiz, 2020-12-31:
Ok, so i don't know if we need to open another issue... YI Home "9FUS" "Y31" works with hack y20ga and 8.2.0.0A_201912270941 firmware ( at this stage you can't upgrade firmware, from app don't work... I have to unhack and then upgrade )... Now with 8.2.0.0A_202012041528 firmware the hack y20ga works BUT when you restart, the cam don't get connection anymore ( Maybe the same problem had @chl3b ). Anyone can confirm this? Actually i unhack again and camera works normally.

P.D. Happy New Year for everyone :-)

risturiz, 2021-01-06:
Hi again... I tested the hack with another microsd brand and now works with last firmware... So the problem is solved for me :-)

roleoroleo, 2021-01-08:
Problem impossible to find...

Mafiewww, 2021-01-10:
Hi roleoroleo. I have the same issue. My camera cant connect to.Wi-Fi repeating wifi password error...
I test with other Wi-fi and same issue...
This is my file : https://drive.google.com/drive/folders/17N4Cznd8lpujdhoKImXv_wzSjT27J2cO?usp=sharing

Do you think that restore.tar.gz is ok?

roleoroleo, 2021-01-14:
If your cam is not bricked and you want to restore the factory fw, simply follow this wiki page: https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack
If you want to try to set wifi user and password use this: https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Change-WiFi-credentials

Mafiewww, 2021-01-14:
Hi,

I try to unhack my camera but when i want to set wifi, i always had an error...
I try with a hotspot (mobile) or with my Xiaomi hotspot.

roleoroleo, 2021-01-16:
After you restored the factory environment, are you able to set wifi like an original cam?
Remove the sd when you do this.

Mafiewww, 2021-01-17:
Hi,

After unhack my camera, remove my sd card and reset my camera with a pin.
I have again a " wifi Password error".

Thank for you help

bagira2005, 2021-09-03:
Hi all . brought a dead body. when connecting .... here is the data              [38]HELLO! BOOT0 is starting!
[41]BOOT0 commit : 95377ec
[44]board init start
[46]set pll start
[48]set pll end
[50][pmu]: bus read error
[52][pmu]: bus read error
[55]board init ok
[56]chip id check OK
[58]DRAM BOOT DRIVE INFO: V0.41
[62]DRAM CLK = 528 MHz
[64]DRAM Type = 2 (2:DDR2,3:DDR3)
[67]DRAMC read ODT  off.
[69]DRAM ODT off.
[72]DRAM SIZE =64 M
[79]DRAM simple test OK.
[81]rtc standby flag is 0x0, super standby flag is 0x0
[86]dram size =64
[89]spinor id is: 52 22 17, read cmd: 0b
[93]Succeed in reading toc file head.
[97]The size of toc is 4c000.
[185]Entry_name        = optee
[188]Entry_name        = u-boot
[192]Entry_name        = dtb
[195]Jump to second Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-27-g7f6d3375 #1 Thu D                                                                                                                                                              ec 10 03:28:16 UTC 2020 arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:127: no pmu node

bagira2005, 2021-09-03:
who has any thoughts ........ for recovery

roleoroleo, 2021-09-03:
It seems an error accessing to the device tree.
Probably you have to restore some partitions.
There is a wiki about the restore of the home partition.
You could use it as a template.

bagira2005, 2021-09-04:
Thanks.
there is a second camera of this series Live, I can take everything there.
USB UART is
tell me the instructions? or maybe someone did already read

roleoroleo, 2021-09-10:
To backup the flash, you can use the sd card method.
Only the restore when the cam is not accessible requires the uart.

closed 2024-05-13 by github-actions[bot]