roleoroleo/yi-hack-Allwinner #72: Help with corrupt /home/app/init.sh

SecAficionado, 2020-06-30:
Hi there,

My camera works but I cannot connect to it using the network.

I was trying to configure WiFi manually, without using the app (more on that in a future post), but I must have entered a typo for a command, or something happened to the `init.sh` file.

The camera connects to WiFi, but the hack never executes, so I have a camera that does not have httpd, ssh or any other server running. It does not connect to the app either (I had turned off the cloud with the hack). Trying to reinstall the hack does not work. I tried to reinstall the firmware, but it doesn't work either. I think I saw a line of code somewhere that discards the firmware if it is not a higher version than the current firmware. I'm stuck!

Is there a way to force the camera to execute something before `init.sh`? I tried creating a config.sh file containing only this command:
`cp /home/app/init.sh /tmp/sd/init.sh`
I wanted to see if I could identify the issue. But the script never executes.
I hope I can execute a script that copies a clean version of init.sh on top of the bad one.

I have also tried resetting the camera, but that only makes it forget the WiFi password. Everything else is the same.

IssisX, 2020-06-30:
I believe equip_test.sh in a folder named "test" on root of sd card would be your answer as it was mine

SecAficionado, 2020-06-30:
Thank you @IssisX 
I tried "test" and "Test", but no luck. The camera does not execute the scripts at all.
I do get logs in the /tmp/sd/log folder, so the SD card is mounted and can be accessed by the camera, but the scripts in it are not executed.

BTW, I had a working version of the WiFi script, but I ran into trouble when I was trying to make sure it survived reboots. Basically, it uses custom `wpa_supplicant.conf` and `resolv.conf` files, and overrides the original `wifi.sh` and `default.script` files to connect to the Internet without the Chinese DNS servers.

The script works once the camera is connected to WiFi, but I want to get it to work when the camera has never been configured for WiFi. All you would have to do is set your SSID and password on the `wpa_supplicant.conf` file.

SecAficionado, 2020-07-01:
I was able to update the firmware: Downloaded the official firmware, copied it to an SD card, renamed it home_y20gam, and inserted it in the camera.

It seemed to do its thing, but the problem persists. I did notice that the firmware was renamed and the .done part was added to the name. My guess is that they do not touch the `init.sh` script to retain WiFi settings and config.

One possibility is that if I create new firmware _with_ an `init.sh` file, it will overwrite the broken one. Can someone please post a clean `init.sh`?

roleoroleo, 2020-07-01:
See this post:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/69#issuecomment-651395650

SecAficionado, 2020-07-01:
Thank you, @roleoroleo!
I'll get to work on this.

You totally called it in your (https://github.com/roleoroleo/yi-hack-Allwinner/issues/69#issuecomment-651239695):
> * /home/app/init.sh
> * /home/base/tools/extpkg.sh
> 
> If you make a mistake editing the last two files the cams could brick.

SecAficionado, 2020-07-02:
Thanks again for your help. Turns out I managed to enter a typo just before the line that calls `/tmp/sd/Factory/config.sh`, so I prevented it from ever working.

I ended up having to use the serial port (UART). But the good part is that I didn't have to flash a new firmware. I was able to fix the script directly and everything started working again. Thanks for the tutorial on backing up using serial. I would only add the speed settings for the serial connection. It should be 115200.

BTW, my wifi script works as far as the network connections are concerned, but there is something that happens when the camera first connects to the Yi cloud (or something in the cloud API) that enables it. Otherwise, applying my hack to a new camera connects to the local wifi, turns on all the servers, but the camera refuses to take snapshots or serve streams. This is a completely different topic, so I should probably open a new issue for this.

I am not sure if you knew this, but the camera sends a request with the wifi ssid and password to the cloud when it is activated. I suspected that, and that's why I did not want to use their cloud at all. I haven't checked that end, but I think the app does the same thing too. It refuses to create a QR code unless it is connected to wifi.

This is a sanitized sample from the camera log:
`https://<their cloud>/v4/ipc/on_line?hmac=[camera HMAC]]&seq=9&uid=[encrypted user id]]&password=[encrypted pwd]]&version=[fw ver]]&model=0&port=0&mac=[wifi mac]]&packetloss=0&p2pconnect=0&p2pconnect_success=0&tfstat=10000&timestamp=[]]&ext_info={"p2p_encrypt":"true","ssid":"your ssid","mac":"your camera mac","ip":"camera ip","signal_quality":"54","powerstate":"0"}`

I'm not sure why they send the camera mac several times, but they capture lots of information about each user. The Android app, for example, requires precise location to share wifi info, so they have that too.

closed 2020-07-05 by SecAficionado