roleoroleo/yi-hack-Allwinner #73: camhash?  No RTSP tab

Merlynda, 2020-07-01:
Yi home 1080p BFUSY311P4ZF3X200306. Yi hack allwinner worked. I can get to web interface configuration. But there is no RTSP tab. I have RTSP server radio button enabled. 
How do I get camhash so I can donate and stream RTSP? Thank you

roleoroleo, 2020-07-01:
You don't need to donate.
rtsp implementation is open source.
Simply try and use it.

But if you want to donate follow the link at the bottom of the main page. :)

Merlynda, 2020-07-01:
Hummm I typed 

rtsp://MyCamIP/ch0_0.h264 

Into my browser and it worked. 

Plus I used http://MyCamIP/8080/cgi-bin/service.sh?name=RTSP&action=START&param1=HIGH

As you can tell I don’t have a clue what I am doing. 

Now I am going to try to get it to work with

https://channelstore.roku.com/details/143683/ip-camera-viewer-pro

roleoroleo, 2020-07-01:
> Plus I used http://MyCamIP/8080/cgi-bin/service.sh?name=RTSP&action=START&param1=HIGH

This step is not needed.

> As you can tell I don’t have a clue what I am doing. 
> 
> Now I am going to try to get it to work with
> 
> https://channelstore.roku.com/details/143683/ip-camera-viewer-pro

Does this sw support onvif?

Merlynda, 2020-07-01:
Yes the roku app does

“Developed by:
Ioan Ghip   

IP Camera Viewer Pro allows you to view live video from most IP cameras on your TV.

It fully supports Motion-JPEG streams, RTSP (jpg encoded), ONVIF, Motion Detection, PTZ for major brands cameras and Basic/Digest Authentication. 

You can also see multiple cameras - 2, 3 or 4 - in a Split Screen View.

IP Camera Viewer Pro also provides a screen saver that will display the stream from your favorite cameras in a rectangle that moves around on your TV screen on a black background.”

Merlynda, 2020-07-01:
[image]
[image]

Merlynda, 2020-07-01:
[image]

roleoroleo, 2020-07-01:
I think you should try "Scan" and enter the IP address of the cam.
Don't use snapshot as a stream.

Merlynda, 2020-07-01:
I did and it filled in snapshot URL

P.S. not working in Roku yet...

roleoroleo, 2020-07-01:
Probably Roku supports only rtsp with jpeg encoding.
This is not common, normally rtsp uses h264 or h265 encoding.
Yi cam uses h264 and it doesn't depends on my hack. I can't change this behavior.

Merlynda, 2020-07-01:
I think you are correct regarding roku only supporting rtsp with jpeg. 

I sent developer the question

closed 2020-07-14 by roleoroleo