roleoroleo/yi-hack-Allwinner #75: Question: Where to start

jonferreira, 2020-07-02:
First of all, thanks for your work.

Secondly, I must admit I feel like a complete dumbass noob making this question but...where to start?

I just received a 4 pack of 1080p Yi Home cameras. According to the QR code they're all BFUS***

I haven't even powered them on, literally just opened the box to see the version. Am I supposed to turn them on, add them to some sort of Yi App to connect to my home wifi, update fw etc and *then* install the hack or can I just dive in and install the hack straight away?

If I can / should install the hack straight away how / where do I then add my home wifi SSID & Password?

My end goal is to use them with Synology Surveillance Station via ONVIF and MQTT for motion alerts.

Thanks

roleoroleo, 2020-07-02:
Configure your cam with the official app and then apply the hack.

jonferreira, 2020-07-02:
dang that was quick!

Thanks will do. Probably worth o use your "backup hack" as well.

Again, thank you!

Merlynda, 2020-07-05:
Please donate to roleo. He reserves at least a beer or two!

roleoroleo, 2020-07-05:
> Thanks will do. Probably worth o use your "backup hack" as well.

Yes, making a backup is always a good thing.

jonferreira, 2020-07-06:
If this works as I want it to, I'm happy to offer a 6 pack

jonferreira, 2020-07-06:
feeling a bit dumb here!

I've downloaded the backup (SD) script, upload the Factory folder (I've first tried putting the 2 files on the SD card root but nothing happens),waited for 30 min but even though the blue LED is solid I can't connect to the camera. Ended up shutting it down to check the SD card but no /backup folder was created.

Am I doing something wrong?

roleoroleo, 2020-07-06:
Yes, you have to copy the Factory folder.
Pay attention to the uppercase "F".
If it doesn't work, try another sd card.

Check the yi fw version inside the app.

jonferreira, 2020-07-06:
Thank you @roleoroleo 

Tried with different SD card already, card was formatted via the app as well.

Current camera fw is 4.5.0.0C_201910080934

roleoroleo, 2020-07-06:
Your cam is not Allwinner but MStar based.
See the proper hack.

jonferreira, 2020-07-06:
Yes just realised I should be looking at https://github.com/roleoroleo/yi-hack-MStar instead,sorry about that.

Doesnt' look like there's a way do dump the stock fw via SD card on that one though?

roleoroleo, 2020-07-06:
Probably there is a similar method.
But I didn't find it.

jonferreira, 2020-07-06:
yeah couldn't find it either :(

guess I need to make a decision then, either take my chances flashing https://github.com/roleoroleo/yi-hack-MStar with no backup or just tear it down and try to connect via serial

closed 2020-11-21 by roleoroleo