roleoroleo/yi-hack-Allwinner #76: MQTT isn't working

Dragon0501, 2020-07-02:
After installation the first setup was made. in the broker are two objects, but there is no motion detection. The states doesn't change...

roleoroleo, 2020-07-03:
Check the client id. It must be unique in your mqtt network.
Check the topic suffix for motion detection.

Dragon0501, 2020-07-03:
The client ID is unique. I have just one cam in my whole network. I don't know what you mean by check the topic suffix. What do I have to check?

Dragon0501, 2020-07-03:
[image]

Dragon0501, 2020-07-03:
Thats the view of the broker. But it doesn't refresh...

Dragon0501, 2020-07-03:
[image]
[image]
Config of the broker... The User and password must be correct, because the connects to the broker.

roleoroleo, 2020-07-03:
I mean this:
[image]

Dragon0501, 2020-07-03:
I changed it. 
Topic Suffix is motion
The other ones are just start and stop. If it is necessary to keep the original ones I can change it again, but it doesn't works with the original ones too.

Dragon0501, 2020-07-03:
Is ist important to keep the original Topics?

Dragon0501, 2020-07-03:
[image]
Reset to former values
Doesn't help, still not working.

roleoroleo, 2020-07-03:
It's not necessary to restore defaults.
You can change it as you want.

Try to read this issue:
https://github.com/roleoroleo/yi-hack-MStar/issues/189

Dragon0501, 2020-07-03:
I read it. Do you want to tell me to reset the camera or what do you want to tell me? How do I flash an older version if you want me to do that. I don't understand what you want to tell me... I am sorry. I think the problem is the same I have...

Dragon0501, 2020-07-03:
Thank you very much! That solved the problem! First flash the 0.1.5 after that update through the software and worked at first try!!!! Amazing! Thank you for your assistance!

closed 2020-07-03 by Dragon0501