roleoroleo/yi-hack-Allwinner #77: RTSP Problem

Dragon0501, 2020-07-03:
My MQTT didn't work. The problem were solved by installing an older version and update it from the website. The MQTT works since this update. Now it is impossible to switch to the low resolution for the tzsp stream. High Res does work. If i switch to the low resolution in the config the stream isn't reachable. 
http://ip-cam:8080/cgi-bin/snapshot.sh works!
MQTT works!
rtsp://IP-CAM:8554/ch0_1.h264 doesn't work!
rtsp://IP-CAM/ch0_0.h264     works with the high res. stream.

roleoroleo, 2020-07-03:
I am in the middle of a breaking change with the RTSP.
Try rtsp://IP-CAM/ch0_1.h264

Dragon0501, 2020-07-03:
Doesn't work... I am sorry

Dragon0501, 2020-07-03:
Perfect! Thank you very much!

closed 2020-07-03 by Dragon0501

roleoroleo, 2020-07-03:
Did you solve the problem?

Dragon0501, 2020-07-03:
You solved it! The changed URL worked!