roleoroleo/yi-hack-Allwinner #78: model 9FUS totally bricked

gjunger, 2020-07-04:
after hacking the 9FUS, firmware 8.2.0 with 0.1.6 (the cam rebooted once and solid blue light) the cam is totally dead. Can´t restart anymore. Also tried to unhack. Nothing happens. 
Is there a chance to get the cam back to life again?
Thank you for any idea I can try

roleoroleo, 2020-07-05:
If the led is solid blue, the cam should work.
Open the yi app and try to use it.

gjunger, 2020-07-13:
the led was solid blue during the hacking process. but never seen again after rebooting the cam. cam is not usable and not recognised by the yi app. so I think I can trash it :-(

roleoroleo, 2020-07-13:
Try to login with telnet.

gjunger, 2020-07-19:
Sorry if I´m boring you, but the cam is not connected to the network, I can´t find any IP adress i could use for telnet. The cam does not react to anything!

roleoroleo, 2020-07-19:
You could try to recover it following the unbrick guide.
See the wiki.

gjunger, 2020-07-20:
Thank you very much for your support. But it´s not worth to waste our time for this 18€ camera. I´ll throw it out of the window! But I have two more cameras and I am afraid to use your great hack again. The only reason for me is to get ftp access. Thx and greets from Austria

roleoroleo, 2020-07-20:
If you have to change idea let me know.

closed 2020-11-21 by roleoroleo